zephyrproject-rtos/zephyr #11649: net: ipv4: Zephyr drops ethernet frame with a trailer

rmstoi, 2018-11-26:
Zephyr drops packet with an ethernet frame with a trailer (thus, whole packet is over 1500 octets). Linux replies correctly to this packet, as seem in attached pcap.

[ipv4-eth-trail-zephyr.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2615144/ipv4-eth-trail-zephyr.pcap.gz)
[ipv4-eth-trail-linux.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2615145/ipv4-eth-trail-linux.pcap.gz)

pfalcon, 2018-11-26:
I believe I was hit by this lately while developing a driver for new eth hardware. I'm not sure, because it's not clear what "ethernet frame with a trailer" is. In my case, HW returned frame with FCS and padded to uint32_t, and it was fed in Zephyr, and what I got from it is a confusing log message and no explicit indication that the packet is dropped.

pfalcon, 2018-11-26:
(That's why I'm -1 on any changes which drop packets without logging.)