zephyrproject-rtos/zephyr #11651: Time consumption is not constant during the pend and unpend operation with 0(1) pending queue

ajaykish, 2018-11-26:
Time consumption is not constant during the pend and unpend operation with pending queue
while WAITQ_SCALABLE Configuration enabled.
As per expectation operations should happen in constant time to meet the O(1)
time complexity. 
Measurement done with the help of #11650 

Steps to reproduce the behavior:
cd tests/kernel/pipe/pipe_api/
mkdir build
cd build
cmake -D BOARD=qemu_x86 -GNinja ../
ninja run

Console Output:
starting test - test_bench
time spent during pending for thread1  444114
time spent during pending for thread2  90650
time spent during pending for thread3  102716
time spent during pending for thread4  34068
time spent during unpend of thread 214068
time spent during unpend of thread 4808
time spent during unpend of thread 50190
time spent during unpend of thread 28642
PASS - test_bench

**Additional context**
Tried to measure the time while CONFIG_SCHED_MULTIQ is enabled and we get the below values.
time spent during pending for thread1  466
time spent during pending for thread2  415
time spent during pending for thread3  130
time spent during pending for thread4  65
time spent during unpend of thread 1104
time spent during unpend of thread 16
time spent during unpend of thread 16
time spent during unpend of thread 16
PASS - test_bench

Unpend operation is constant while CONFIG_SCHED_MULTIQ is enabled.

pfalcon, 2018-11-27:
@andyross, @andrewboie: FYI.

andyross, 2018-11-27:
WAITQ_SCALABLE is O(log2(N)) in the number of threads.  MULTIQ is constant time (but has a very strict limit on the number of priorities).  DUMB is O(N).

The scatter seems very high though, I'll take a look.  Note that some of the MULTIQ numbers seem inapprioriately fast -- I really doubt you can get a thread unpended in 16 cycles even if it's a simple list remove.

andyross, 2018-11-27:
OK, took a look at the submitted code, but didn't get a chance to run it.

This isn't really testing anything useful.  When someone says "O(N)"
or "O(1)" or whatever, they're talking about scaling behavior as the
number of objects (suspended/active threads in this case) changes.

This test just tries four threads.  If you want to measure scaling,
you have to do it with dozens to thousands.

Also, it's hitting an inverse scaling problem.  Every time it pends a
thread it removes it from the scheduler run queue but **adds it** to a
waitq.  So as your scheduler should be getting faster your waitq is
getting slower, and you don't know what you're measuring.

What I'm getting looking at the numbers above:

1. There's way too much jitter.  These operations should be relatively deterministic (though the rbtree backend of SCALABLE has different code paths and won't be exactly the same every time).

2. The rbtree numbers seem shockingly high (they should be in the hudreds-to-low-thousands of instructions), and the multiq numbers seem too low in a few cases.

Can you repeat the test on real hardware?  I'm wondering if our newer qemu version is a little bouncy when looking at the TSC (which it tries to slave to real host time via various tricks).