zephyrproject-rtos/zephyr #11659: Network logging intrusive behavior

rlubos, 2018-11-26:
Currently, network logger helpers (including logger module registration) are defined in `net/net_core.h`. This file is included implicitly by multiple other networking headers (for instance `net/net_app.h` or `/net/socket.h`. 

This results in quite intrusive behavior of networking headers which expect that a module including them is already a user of the logger with defined `LOG_LEVEL`. Otherwise `net/net_core.h` will register a new logger module on it's own, with undefined `LOG_MODULE_NAME`. The problem with that is, that in the latter case, we will get linker errors, if it happens in two or more modules:
```
/home/rolu/repos/zephyr/include/net/net_core.h:55: multiple definition of `log_const_LOG_MODULE_NAME'
```

We should prevent this situation by either:
1) Providing more strict preprocessor checks in `net/net_core.h`, for instance verify that a proper `LOG_MODULE_NAME` was defined before registering new logger module,
2) Move logging helpers into a separate header, that shall only be included explicitly in source files, after specifying logger parameters, for instance:
```
#define LOG_MODULE_NAME net_app
#define NET_LOG_LEVEL CONFIG_NET_APP_LOG_LEVEL
#include <net/net_log.h>
```

Personally I'd go with second solution and provide a separate header for logger helpers, but I wanted to hear from others before actually fixing the issue.

jukkar, 2018-11-28:
The net_core.h will call `LOG_MODULE_REGISTER(LOG_MODULE_NAME);` only if NET_LOG_LEVEL is set, and it should be set only in one .c file before including net_core.h.
Can you point to sources which shows the error you mentioned?

rlubos, 2018-11-28:
> The net_core.h will call LOG_MODULE_REGISTER(LOG_MODULE_NAME); only if NET_LOG_LEVEL is set, and it should be set only in one .c file before including net_core.h.

The problem is it isn't. In case when neither `LOG_LEVEL` nor `NET_LOG_LEVEL` are defined, we get:
https://github.com/zephyrproject-rtos/zephyr/blob/c43a116f4182e8848b55a7cd2d7788cc5a4f6338/include/net/net_core.h#L40

Which will make the following code in the header to register the logger module.

> Can you point to sources which shows the error you mentioned?

I don't have any code that I can share publicly at this point, yet the behavior described is fairly easy to trigger. Let's just take for example `sntp_client` sample that do not use LOG nor NET_LOG at the moment, and try to use the logger, by registering it just below header inclusions:
```
#include <logging/log.h>
LOG_MODULE_REGISTER(app);
```
Doing this will trigger bunch of errors during compilation, as the logger module was already registered in `net_core.h` included implicitly.

jukkar, 2018-11-29:
I am working on this and will try to figure out easy way to solve the issue. This issue is basically the same as #11343

commit 86689030 (2018-12-07): net: Clarify logging in networking code

Remove network specific default and max log level setting
and start to use the zephyr logging values for those.

Remove LOG_MODULE_REGISTER() from net_core.h and place the
calls into .c files. This is done in order to avoid weird
compiler errors in some cases and to make the code look similar
as other subsystems.

Fixes #11343
Fixes #11659
closed 2018-12-07 by jukkar