zephyrproject-rtos/zephyr #11670: ci: unlicense misrated as non-permissive

b0661, 2018-11-27:
**Describe the bug**

PR #10885 contains a file that is licensed **"unlicense"** ( see https://spdx.org/licenses/Unlicense.html). This is a much more permissive license than apache-2.0.  It should not make CI fail.

**To Reproduce**

Add a file wirh "SPDX-License-Identifier: Unlicense" and run CI.

**Expected behavior**

CI license verification shall pass on  "SPDX-License-Identifier: Unlicense".

**Impact**

CI of PR #10885, #10888  declared failed.

**Screenshots or console output**

PR #10885

**Environment (please complete the following information):**

Zephyr CI

**Additional context**

nashif, 2018-12-08:
Any file going into the zephyr tree with a different license that the project license needs to be reviewed. This check in CI is currently not 'required' for merging and still under heavy development, so a maintainer can override this check after reviewing the warnings and determining that the files are ok.

Will add some text explaining the above in the comment posted by CI.

nashif, 2019-01-08:
We now add more information about the license scan, any license warning need to be reviewed and processed by someone, making them PASS is just going to be misleading. Those who merge can always override the check. My concern is that if we make it PASS, nobody will look at the warnings of non APL licensed files we are pulling into the tree.

The tool we are using claims the 'unlicense' license is not permssive. I need to check what is going on.

nashif, 2019-01-17:
@kestewart Any idea why "unlicense" is showing as non-permissive in scanacode-toolkit?