zephyrproject-rtos/zephyr #11678: newlib pow function broken for galileo

laperie, 2018-11-27:
**Describe the bug**
Calling newlib's pow function consistently crashes Galileo
following call always fails:  pow(2,-3);

**To Reproduce**
Steps to reproduce the behavior:
1. Hello world sample
2. Apply diff  [pow-hello-world-diff.txt](https://github.com/zephyrproject-rtos/zephyr/files/2619600/pow-hello-world-diff.txt)
3. Build and flash for Galileo
4. Observe console output

**Expected behavior**
It should work

**Impact**
pow function of newlib cannot be used. 

**Screenshots or console output**

`***** Booting Zephyr OS zephyr-v1.13.0-2183-gdbf405ab21 *****
Hello World! galileo
***** CPU exception 7
Current thread ID = 0x0010b4e0
eax: 0x0010809c, ebx: 0x00000000, ecx: 0x9000b020, edx: 0x00108094
esi: 0x00000000, edi: 0x00000000, ebp: 0x0010ba90, esp: 0x0010ba00
eflags: 0x00210206 cs: 0x0008
eip: 0x00100323
Fatal fault in essential thread! Spinning...`

**Environment (please complete the following information):**
Zephyr master as of dbf405ab218e9a104ff7af490551cedda38b738a 
Zephyr SDK 0.9.5
Linux

andrewboie, 2018-11-27:
CPU Exception 7 happens when a floating point operation is enabled, but the FPU isn't activated. The pow() function requires the FPU internally, its args are doubles.

Is CONFIG_FLOAT turned on?
Also, if more than one thread uses the FPU, you'll also need to enable CONFIG_FP_SHARING.

laperie, 2018-11-30:
@andrewboie  you were absolutely right! FPU was not activated for Galileo (in a GPTP example). I wonder if we need to enforce it whenever NEWLIB is enabled..   But closing this issue for now.

closed 2018-11-30 by laperie