zephyrproject-rtos/zephyr #11683: Invalid hyperlink in some README pages

jsplyy, 2018-11-27:
**Invalid hyperlink in some README pages**

**To Reproduce**
Steps to reproduce the behavior:
1. https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/bluetooth/hci_uart#id1
2. https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/bluetooth/beacon

**Expected behavior**
link is invalid in these read pages.
[image]

carlescufi, 2018-11-27:
This is not an invalid link. We use `rst` to format documentation but GitHub doesn't render it properly. Instead of browsing it on GitHub, you should browse it on the documentation website:

https://docs.zephyrproject.org/latest/samples/bluetooth/beacon/README.html

carlescufi, 2018-11-27:
@jsplyy thanks for reporting, I am planning to add a FAQ in the docs and I will include this.

closed 2018-11-27 by carlescufi

jsplyy, 2018-11-27:
Thanks a lot. @carlescufi