zephyrproject-rtos/zephyr #11698: net: ipv4: Zephyr replies to ICMPv4 packets with incorrect checksums

rmstoi, 2018-11-28:
These packets should be dropped instead. Note that also checksum 0 is not valid for ICMP.

[icmpv4-bad-chksum2.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2623574/icmpv4-bad-chksum2.pcap.gz)
[icmpv4-bad-chksum1.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2623575/icmpv4-bad-chksum1.pcap.gz)

rlubos, 2018-12-13:
@jukkar Looks like this one was fixed in #11839, specifically with https://github.com/zephyrproject-rtos/zephyr/commit/6c291524bc38e038f62355b9cb87a7140beda5a0. I am no longer able to reproduce this (which makes sense because the fix seems very reasonable). I think we could close this one.

rmstoi, 2018-12-13:
@rlubos You are right, that PR fixed this issue. Thanks!

closed 2018-12-13 by rmstoi