zephyrproject-rtos/zephyr #11702: Add support for nrfx i2s driver

m-nilsson, 2018-11-28:
We would like to use the nrfx i2s driver. Is it possible at this time? If not, is it scheduled to be integrated in the near future?

m-nilsson, 2018-12-03:
I have managed to use the nrfx i2s driver with zephyr but I am still curious to know if it is planned to be included in a future release.
And it would be nice if the higher level i2s api in zephyr would support interrupt driven i2s with a callback fired when more data is needed, like the nordic driver.

m-nilsson, 2018-12-03:
Sounds great, thanks!

jarz-nordic, 2018-12-03:
@m-nilsson : 
@barsok told me he needs to do some clean up on his branch. He should be able to share it with you around Wednesday this week.

hchaudhary1, 2019-03-25:
Currently have to manually add NRFX_I2S_ENABLED as well as add "zephyr_sources(nrfx/drivers/src/nrfx_i2s.c)"

carlescufi, 2019-03-25:
@jarz-nordic @anangl any news on this front?

binkboyd, 2019-08-20:
@jakub-uC , @barsok - I am also looking at using the nrfx I2S driver in zephyr; would it be possible to get access to the in-progress branch ? I would much prefer to get started with something that has been tested some rather than start from scratch myself .)

jakub-uC, 2019-08-20:
@binkboyd : I am affraid that temporary branch has not been updated for a long time. As a result it might be hard to sync with current zephyr master.
Personally I recommend to use directly nrfx driver without I2S shim.

@barsok : Please correct me if I am wrong.

barsok, 2019-08-20:
@jakub-uC @binkboyd  hi!  We have a working SHIM and is now being prepared for review. @adamkondraciuk would you please comment?

adamkondraciuk, 2019-08-20:
Hi. The code needs to be cleaned and refactored. There are many comments, useless logs and awkward test solutions.

barsok, 2019-08-20:
@jakub-uC @binkboyd  please let us know if you would like to work with dirty but working code or if you prefer to wait until we put a PR to github (soon)?

jakub-uC, 2019-08-20:
@barsok : My impression was that @binkboyd can leave with dirty code.

binkboyd, 2019-08-20:
hi, I found https://github.com/barsok/zephyr/tree/bsokol-i2c which I've merged into zephyr master manually. Is this the current "dirty" code ? I'm still working on some other hardware, so I am yet to test other than that it compiles. @barsok - what is the timeline on the PR ?

adamkondraciuk, 2019-08-20:
Hi @binkboyd . This code is not up to date. Tomorrow I will share in my fork current I2S driver code.

binkboyd, 2019-08-21:
excellent; thanks!

binkboyd, 2019-08-21:
@adamkondraciuk ; do you have an ETA? I was testing with the older code from @barsok  today and was not able to get it to work. Dirty code is no problem as long as it works :)

adamkondraciuk, 2019-08-22:
Hi. Sorry for lag - I had some technical problems. Here is the driver: https://github.com/adamkondraciuk/zephyr/tree/i2s_test

binkboyd, 2019-08-22:
hm; couldn't find anything there - are you sure you pushed the code to github ?

adamkondraciuk, 2019-08-22:
sorry - now it should be OK

binkboyd, 2019-08-22:
excellent; thanks! :+1:

solsbarry, 2020-08-04:
Is there any update on this? Or is there any newer version of the "dirty" code. Thanks.

PR #34251 (2021-07-28): drivers: i2s: Add support for nRF I2S peripherals
Add a shim that allows using the nrfx I2S driver via the Zephyr API.
Add also missing devicetree nodes representing the I2S peripherals
in the nRF52 Series SoCs.

Bases on #34250.

Resolves #11702.
closed 2021-07-28 by cfriedt