zephyrproject-rtos/zephyr #11712: Add support for newlib-nano

oxymoron87, 2018-11-28:
**Is your feature request related to a problem? Please describe.**
Can't opt for linking with newlib-nano when choosing "(top menu) -> C Library -> Build with newlib c library"

**Describe the solution you'd like**
Ability to set newlib-nano option can dramatically decrease the size of project (in my case it is small C++ project with use of std::vector)
~~~~
newlib
Memory region         Used Size  Region Size  %age Used
            FLASH:      101844 B         1 MB      9.71%
             SRAM:       11336 B       256 KB      4.32%
         IDT_LIST:          56 B         2 KB      2.73%
~~~~

~~~~
newlib-nano
Memory region         Used Size  Region Size  %age Used
            FLASH:       32960 B         1 MB      3.14%
             SRAM:        9188 B       256 KB      3.50%
         IDT_LIST:          56 B         2 KB      2.73%
~~~~
For now I've added **--specs=nano.specs** to CMakeLists.txt:
~~~~
if(NOT CONFIG_NATIVE_APPLICATION)
zephyr_ld_options(
  -nostdlib
  -static
  -no-pie
  --specs=nano.specs
)
endif()
~~~~

carlescufi, 2018-12-07:
@oxymoron87 note that there is an ongoing discussion about the C library and and types that @galak is leading.

galak, 2019-09-04:
I think we support this now.

closed 2019-09-04 by galak