zephyrproject-rtos/zephyr #11714: Building libsodium with arm-zephyr-eabi

sijohans, 2018-11-28:
Hello,

I am trying to build libsodium (https://github.com/jedisct1/libsodium) to a zephyr project. However, i fail to configure it with arm-zephyr-eabi-gcc. I followed the example **samples/application_development/external_lib** and tried to add it as an external project. I manage to build it for the native_posix target but fails to build it for nrf52840_pca10056.

This is how i made it work for the posix target:
```cmake
include(ExternalProject)
set(sodium_src_dir   ${CMAKE_CURRENT_SOURCE_DIR}/libsodium)
set(sodium_build_dir ${CMAKE_CURRENT_BINARY_DIR}/libsodium)
set(LIBSODIUM_LIB_DIR     ${sodium_build_dir}/lib)
set(LIBSODIUM_INCLUDE_DIR ${sodium_src_dir}/src/libsodium/include)
ExternalProject_Add(
    sodium_project
    PREFIX     ${sodium_build_dir}
    SOURCE_DIR ${sodium_src_dir}
    BINARY_DIR ${sodium_src_dir}
    CONFIGURE_COMMAND ${sodium_src_dir}/autogen.sh
    BUILD_COMMAND
    ${sodium_src_dir}/configure
    --prefix=${sodium_build_dir}
    CC=${CMAKE_C_COMPILER}
    AR=${CMAKE_AR}
    CFLAGS=${CMAKE_C_FLAGS}
    INSTALL_COMMAND make -j8 install
    BUILD_BYPRODUCTS ${LIBSODIUM_LIB_DIR}/libsodium.a
)

add_library(sodium_lib STATIC IMPORTED GLOBAL)
add_dependencies(sodium_lib sodium_project)
target_link_libraries(app PUBLIC sodium_lib)
```

I also tried follow the example:
```cmake
zephyr_get_include_directories_for_lang_as_string(       C includes)
zephyr_get_system_include_directories_for_lang_as_string(C system_includes)
zephyr_get_compile_definitions_for_lang_as_string(       C definitions)
zephyr_get_compile_options_for_lang_as_string(           C options)
set(external_project_cflags "${options} ${definitions} ${includes} ${system_includes}")

include(ExternalProject)
set(sodium_src_dir   ${CMAKE_CURRENT_SOURCE_DIR}/libsodium)
set(sodium_build_dir ${CMAKE_CURRENT_BINARY_DIR}/libsodium)
set(LIBSODIUM_LIB_DIR     ${sodium_build_dir}/lib)
set(LIBSODIUM_INCLUDE_DIR ${sodium_src_dir}/src/libsodium/include)
ExternalProject_Add(
    sodium_project
    PREFIX     ${sodium_build_dir}
    SOURCE_DIR ${sodium_src_dir}
    BINARY_DIR ${sodium_src_dir}
    CONFIGURE_COMMAND ${sodium_src_dir}/autogen.sh
    BUILD_COMMAND
    ${sodium_src_dir}/configure
    --prefix=${sodium_build_dir}
    CC=${CMAKE_C_COMPILER}
    AR=${CMAKE_AR}
    CFLAGS=${external_project_cflags}
    INSTALL_COMMAND make -j8 install
    BUILD_BYPRODUCTS ${LIBSODIUM_LIB_DIR}/libsodium.a
)

add_library(sodium_lib STATIC IMPORTED GLOBAL)
add_dependencies(sodium_lib sodium_project)
target_link_libraries(app PUBLIC sodium_lib)
```

If i do this, it will fail in the configure step when detecting if the compiler works. But just using ${CMAKE_C_FLAGS} seems to work.

Building libsodium with arm-none-eabi-gcc works fine and i have been trying a lot of combinations to just configure it uzing arm-zephyr-eabi-gcc but i cant pass the step where it verifies that the  compiler work.

Any ideas or suggestion how to proceed with this?

carlescufi, 2018-11-28:
CC @SebastianBoe

carlescufi, 2018-11-28:
@sijohans since Zephyr also supports standard `arm-none-eabi-gcc`, have you tried switching from `ZEPHYR_TOOLCHAIN_VARIANT=zephyr` to `ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb` as described [here](https://docs.zephyrproject.org/latest/getting_started/toolchain_3rd_party_x_compilers.html#gnu-arm-embedded)?

sijohans, 2018-11-30:
If my cmake script looks like this:
```cmake
zephyr_get_include_directories_for_lang_as_string(       C includes)
zephyr_get_system_include_directories_for_lang_as_string(C system_includes)
zephyr_get_compile_definitions_for_lang_as_string(       C definitions)
zephyr_get_compile_options_for_lang_as_string(           C options)

set(external_project_cflags "${options} ${definitions} ${includes} ${system_includes}")
include(ExternalProject)
set(sodium_src_dir   ${CMAKE_CURRENT_SOURCE_DIR}/libsodium)
set(sodium_build_dir ${CMAKE_CURRENT_BINARY_DIR}/libsodium)

set(LIBSODIUM_LIB_DIR     ${sodium_build_dir}/lib)
set(LIBSODIUM_INCLUDE_DIR ${sodium_src_dir}/src/libsodium/include)

ExternalProject_Add(
    sodium_project
    PREFIX     ${sodium_build_dir}
    SOURCE_DIR ${sodium_src_dir}
    BINARY_DIR ${sodium_src_dir}
    CONFIGURE_COMMAND ${sodium_src_dir}/autogen.sh
    BUILD_COMMAND
    ${sodium_src_dir}/configure
    --prefix=${sodium_build_dir}
    CC=${CMAKE_C_COMPILER}
    AR=${CMAKE_AR}
    CFLAGS=${external_project_cflags}
    INSTALL_COMMAND make -j8 install
    BUILD_BYPRODUCTS ${LIBSODIUM_LIB_DIR}/libsodium.a
)
add_library(sodium_lib STATIC IMPORTED GLOBAL)
add_dependencies(sodium_lib sodium_project)
set_target_properties(sodium_lib PROPERTIES IMPORTED_LOCATION             ${LIBSODIUM_LIB_DIR}/libsodium.a)
set_target_properties(sodium_lib PROPERTIES INTERFACE_INCLUDE_DIRECTORIES ${LIBSODIUM_INCLUDE_DIR})
add_library(sodium_wrapper src/libsodium_wrapper.c)
target_link_libraries(sodium_wrapper sodium_lib)
target_link_libraries(app PUBLIC sodium_wrapper)
```

Then i run cmake, and then when i try to make it:
```sh
$ make
Scanning dependencies of target sodium_project
[  1%] Creating directories for 'sodium_project'
[  1%] No download step for 'sodium_project'
[  2%] No patch step for 'sodium_project'
[  3%] No update step for 'sodium_project'
[  4%] Performing configure step for 'sodium_project'
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize: copying file 'build-aux/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoconf --force
autoreconf: configure.ac: not using Autoheader
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:76: installing 'build-aux/compile'
configure.ac:10: installing 'build-aux/missing'
src/libsodium/Makefile.am: installing 'build-aux/depcomp'
autoreconf: Leaving directory `.'
[  4%] Performing build step for 'sodium_project'
checking build system type... x86_64-pc-linux-gnu
[... 441 lines cut ...]
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "libsodium"
| #define PACKAGE_TARNAME "libsodium"
| #define PACKAGE_VERSION "1.0.16"
| #define PACKAGE_STRING "libsodium 1.0.16"
| #define PACKAGE_BUGREPORT "https://github.com/jedisct1/libsodium/issues"
| #define PACKAGE_URL "https://github.com/jedisct1/libsodium"
| #define PACKAGE "libsodium"
| #define VERSION "1.0.16"
| /* end confdefs.h.  */
| 
| int
| main ()
| {
| 
|   ;
|   return 0;
| }
configure:3815: error: in `/home/simonj/Work/tmp/zephyr/samples/bluetooth/test/libsodium':
configure:3817: error: C compiler cannot create executables
See `config.log' for more details
```

It seems to be the posix_cheats.h that causes the errors. I'll dig into this also with arm-zephyr-eabi and see what the problem might be.

sijohans, 2018-11-30:
With the arm-zephyr-eabi compiler:
```
configure:3768: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc -Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /home/simonj/Work/tmp/zephyr/samples/bluetooth/test/build-nrf52840/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main --sysroot /opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr -mthumb -mcpu=cortex-m4 -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -Wpointer-arith -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -DKERNEL -D__ZEPHYR__=1 -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -DNRF52840_XXAA -I/home/simonj/Work/tmp/zephyr/kernel/include -I/home/simonj/Work/tmp/zephyr/arch/arm/include -I/home/simonj/Work/tmp/zephyr/soc/arm/nordic_nrf/nrf52 -I/home/simonj/Work/tmp/zephyr/soc/arm/nordic_nrf/nrf52/include -I/home/simonj/Work/tmp/zephyr/soc/arm/nordic_nrf/include -I/home/simonj/Work/tmp/zephyr/include -I/home/simonj/Work/tmp/zephyr/include/drivers -I/home/simonj/Work/tmp/zephyr/samples/bluetooth/test/build-nrf52840/zephyr/include/generated -I/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/include -I/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/include-fixed -I/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include -I/home/simonj/Work/tmp/zephyr/soc/arm/nordic_nrf/include -I/home/simonj/Work/tmp/zephyr/ext/lib/crypto/tinycrypt/include -I/home/simonj/Work/tmp/zephyr/ext/hal/cmsis/Include -I/home/simonj/Work/tmp/zephyr/ext/hal/nordic/nrfx -I/home/simonj/Work/tmp/zephyr/ext/hal/nordic/nrfx/drivers/include -I/home/simonj/Work/tmp/zephyr/ext/hal/nordic/nrfx/hal -I/home/simonj/Work/tmp/zephyr/ext/hal/nordic/nrfx/mdk -I/home/simonj/Work/tmp/zephyr/ext/hal/nordic/. -I/home/simonj/Work/tmp/zephyr/subsys/settings/include -I/home/simonj/Work/tmp/zephyr/subsys/bluetooth -I/home/simonj/Work/tmp/zephyr/subsys/bluetooth -isystem/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/include -isystem/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/include-fixed   conftest.c  >&5
/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/real-ld: cannot find crt0.o: No such file or directory
collect2: error: ld returned 1 exit status
configure:3772: $? = 1
configure:3810: result: no
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "libsodium"
| #define PACKAGE_TARNAME "libsodium"
| #define PACKAGE_VERSION "1.0.16"
| #define PACKAGE_STRING "libsodium 1.0.16"
| #define PACKAGE_BUGREPORT "https://github.com/jedisct1/libsodium/issues"
| #define PACKAGE_URL "https://github.com/jedisct1/libsodium"
| #define PACKAGE "libsodium"
| #define VERSION "1.0.16"
| /* end confdefs.h.  */
| 
| int
| main ()
| {
| 
|   ;
|   return 0;
| }
configure:3815: error: in `/home/simonj/Work/tmp/zephyr/samples/bluetooth/test/libsodium':
configure:3817: error: C compiler cannot create executables
See `config.log' for more details
```
I didn't find any settings like the one i CMake you can use (COMPILER_WORKS)

SebastianBoe, 2018-12-03:
Ah, when it was said ", it will fail in the configure step when detecting if the compiler works. "

I thought you meant when Zephyr tests the compiler, but it fails when the libsodium build system tests the compiler.

I can't comment on how to build libsodium, but if you can figure out what the libsodium build system needs from Zephyr's build system, then I can help you with how to make Zephyr export the necessary metadata to libsodium.

SebastianBoe, 2018-12-28:
Closing as resolved.

closed 2018-12-28 by SebastianBoe