zephyrproject-rtos/zephyr #11719: Legacy MQTT sample app breaking

karthikprabhu17, 2018-11-29:
**Describe the bug**
Whenever I tried to build a mqtt client application with mqtt legacy I get a log const error in linking.

../libapp.a(telit.c.obj):/Users/kprabhuv/new_zephyr/include/net/net_core.h:54: multiple definition of `log_const_LOG_MODULE_NAME'

What have you tried to diagnose or workaround this issue?
I tried using an older zephyr tree and it work. So i did a git bisect and found this commit to break the mqtt sample app.

commit 6a960e7d175c7972666b6dfce911c181f033b52c
Author: Krzysztof Chruscinski <krzysztof.chruscinski@nordicsemi.no>
Date:   Thu Nov 15 07:46:15 2018 +0100

    logging: Fix LOG_LEVEL dependency on log.h include

    Refactoring of LOG_MODULE_REGISTER (commit 88648699) introduced
    regression: fixed ordering of LOG_LEVEL definition and log.h
    include.

    Signed-off-by: Krzysztof Chruscinski <krzysztof.chruscinski@nordicsemi.no>

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. export BOARD=sam_e70_xplained
2. cmake ..
3. make
4. See below error

Memory region         Used Size  Region Size  %age Used
           FLASH:      136296 B         2 MB      6.50%
            SRAM:      109264 B       384 KB     27.79%
        IDT_LIST:          56 B         2 KB      2.73%
../libapp.a(telit.c.obj):/Users/kprabhuv/new_zephyr/include/net/net_core.h:54: multiple definition of `log_const_LOG_MODULE_NAME'
../libapp.a(adapter.c.obj):/Users/kprabhuv/new_zephyr/include/net/net_core.h:54: first defined here
../libapp.a(ehl-oob.c.obj):/Users/kprabhuv/new_zephyr/include/net/net_core.h:54: multiple definition of `log_const_LOG_MODULE_NAME'
../libapp.a(adapter.c.obj):/Users/kprabhuv/new_zephyr/include/net/net_core.h:54: first defined here
../libapp.a(mqtt_client.c.obj):/Users/kprabhuv/new_zephyr/include/net/net_core.h:54: multiple definition of `log_const_LOG_MODULE_NAME'
../libapp.a(adapter.c.obj):/Users/kprabhuv/new_zephyr/include/net/net_core.h:54: first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [zephyr/zephyr_prebuilt.elf] Error 1
make[1]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make: *** [all] Error 2

**Expected behavior**
Successful build

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

This is a showstopper as recent merge to master caused this:

commit 6a960e7d175c7972666b6dfce911c181f033b52c
Author: Krzysztof Chruscinski <krzysztof.chruscinski@nordicsemi.no>
Date:   Thu Nov 15 07:46:15 2018 +0100

    logging: Fix LOG_LEVEL dependency on log.h include

    Refactoring of LOG_MODULE_REGISTER (commit 88648699) introduced
    regression: fixed ordering of LOG_LEVEL definition and log.h
    include.

    Signed-off-by: Krzysztof Chruscinski <krzysztof.chruscinski@nordicsemi.no>

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) = Mac
 - Toolchain (e.g Zephyr SDK, ...) = gnuarmemb
 - Commit SHA or Version used = 0abd92266a2da8150afb8b6af098aae8f6a25cb8

**Additional context**
Add any other context about the problem here.

prj.conf file

  CONFIG_ENTROPY_GENERATOR=y
  CONFIG_HEAP_MEM_POOL_SIZE=16384
  CONFIG_INIT_STACKS=y
  CONFIG_MAIN_STACK_SIZE=2048
  CONFIG_PRINTK=y
  CONFIG_REBOOT=y
  CONFIG_LOG=y
 
  
 # Generic Networking Settings
 CONFIG_NET_IPV4=y
 CONFIG_NETWORKING=y
 CONFIG_NET_TCP=y
 CONFIG_NET_ARP=y
 CONFIG_NET_L2_ETHERNET=y
 CONFIG_NET_LOG=y
 #CONFIG_NET_DEBUG_NET_PKT=n
 #CONFIG_NET_DEBUG_APP=n
 19 #CONFIG_NET_DEBUG_TCP=n
 CONFIG_NET_SHELL=n
 CONFIG_NET_CONFIG_SETTINGS=y
 # CONFIG FOR TELIT
 CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.1.25"
 CONFIG_NET_CONFIG_MY_IPV4_GW="192.168.1.1"
 # For IPv6
 # Enable IPv6 support
 CONFIG_NET_IPV6=n
 CONFIG_NET_BUF_DATA_SIZE=256
 CONFIG_NET_IPV6_RA_RDNSS=y
 CONFIG_NET_CONFIG_PEER_IPV4_ADDR="XX.XXX.XX.XX"
 CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT=3
 
 # Networking TLS Settings
 CONFIG_NET_APP_TLS=y
 CONFIG_MBEDTLS=y
 CONFIG_MBEDTLS_BUILTIN=y
 CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=5000
 CONFIG_MBEDTLS_ENABLE_HEAP=y
 CONFIG_MBEDTLS_HEAP_SIZE=30000
 #CONFIG_MBEDTLS_CFG_FILE="config-mini-tls1_2.h"
 CONFIG_MBEDTLS_DEBUG=n
 CONFIG_MBEDTLS_DEBUG_LEVEL=2
 
 
  # MQTT Settings
 CONFIG_MQTT_LEGACY_LIB=y
 CONFIG_MQTT_LEGACY_LIB=y
 CONFIG_MQTT_LEGACY_LIB_TLS=y
 CONFIG_MQTT_LOG_LEVEL_DEFAULT=y
 CONFIG_MQTT_LEGACY_MSG_MAX_SIZE=1024
 CONFIG_MQTT_LEGACY_SUBSCRIBE_MAX_TOPICS=3
 
 CONFIG_TEST_RANDOM_GENERATOR=y
 CONFIG_TIMER_RANDOM_GENERATOR=y

rlubos, 2018-11-29:
Hi, I've faced the same problem with networking headers, and created a generic issue for that: #11659.

I assume you are trying to use a logger in a source file that includes some networking headers (`net_app.h`, `mqtt_legacy.h` etc.), hence `LOG_MODULE_REGISTER` is called twice. A workaround that works for me is to create logger instance before including any of the headers:
```
#include <logging/log.h>
#define LOG_LEVEL LOG_LEVEL_ERROR
LOG_MODULE_REGISTER(my_module_name);
```
Defining `LOG_LEVEL` will prevent networking headers from creating another logger instance.

In a meantime we should work on fixing this behavior in net headers.

carlescufi, 2018-11-29:
@karthikprabhu17 if #11659 covers the issue describe here please close this one.

karthikprabhu17, 2018-11-29:
@rlubos @carlescufi : thanks for the quick reply. I will follow the generic issue #11659

galak, 2018-12-05:
Closing since we have the generic issue #11659

closed 2018-12-05 by galak