zephyrproject-rtos/zephyr #11731: Mesh friend doesn't forward packets to LPNs non-primary elements

trond-snekvik, 2018-11-29:
The Zephyr BT Mesh friend only forwards packets that 
1. have their destination in the LPN's subscription list 
or 
2. [Are destined for the first element in the LPN.](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/mesh/friend.c#L1193)

This makes the friend queue reject packets that are supposed to go to non-primary elements. The friend module needs to store the LPNs element count in addition to its primary address when the friend request comes in, and take this into account when matching addresses for the LPN.

**To Reproduce**
Make an LPN node with more than one element connect to the friend, and try to send a message from the friend to the LPN's second element. It'll fail [the check on line 1193](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/mesh/friend.c#L1193) and reject the message.

**Expected behavior**
According to section 3.5.5 of the spec, the friend should forward packets destined for all the elements on the LPN node. 

**Impact**
Can't publish to models that aren't in the primary element

jhedberg, 2018-11-29:
That's a really good catch, thanks a lot for reporting it. The current implementation indeed doesn't do anything with the "number of elements" parameter in the Friend Request PDU. This should be too hard to fix. Do you already have a patch, or I should start working on it?

This also looks like a pretty severe oversight in the Test Specification, since I do think we've successfully run all PTS tests for Friendship. Would you mind opening an erratum against the TS once you've confirmed that such a test doesn't exist there?

trond-snekvik, 2018-11-29:
That was fast! 

I hadn't started on a fix, so I'll try the PR.

Looks like there's no test for this, we'll make an erratum.

trond-snekvik, 2018-11-29:
The machine I'm on has an outdated dependency, so I'm blocked until chocolatey has fixed their system outage. 

It's been down for a couple of hours, I'll check back once I have been able to test, but it may have to wait until tomorrow. The changes in the PR should do the trick if the subnet resolution worked out.

trond-snekvik, 2018-11-29:
Was able to run the fix, works great! 👍

PR #11734 (2018-11-29): Bluetooth: Mesh: Fix matching for all elements of an LPN
When we're acting as a Friend for an LPN, we need to consider all
elements of the LPN. The information of how many elements the LPN has
is provided in the Friend Request message, however until now the code
did not do anything with this information.

Fix the issue by tracking the number of elements for each LPN and
update the unicast address matching code to take this into account.

Fixes #11731
commit bec74121 (2018-11-29): Bluetooth: Mesh: Fix matching for all elements of an LPN

When we're acting as a Friend for an LPN, we need to consider all
elements of the LPN. The information of how many elements the LPN has
is provided in the Friend Request message, however until now the code
did not do anything with this information.

Fix the issue by tracking the number of elements for each LPN and
update the unicast address matching code to take this into account.

Fixes #11731
closed 2018-11-29 by jhedberg