zephyrproject-rtos/zephyr #11741: mqtt_publisher sample not working with BLE IPSP

ntavish, 2018-11-29:
**Describe the bug**
MQTT publisher sample is stuck trying to connect over TCP.

Following is after making a change to fix a bug(?) which enables bt twice and quits after `EALREADY` is returned:

```
diff --git a/samples/net/mqtt_publisher/src/main.c b/samples/net/mqtt_publisher/src/main.c
index 64ab8f7afe..175fd5904b 100644
--- a/samples/net/mqtt_publisher/src/main.c
+++ b/samples/net/mqtt_publisher/src/main.c
@@ -448,13 +448,6 @@ static int network_setup(void)
 #if defined(CONFIG_NET_L2_BT)
        const char *progress_mark = "/-\\|";
        int i = 0;
-       int rc;
-
-       rc = bt_enable(NULL);
-       if (rc) {
-               printk("bluetooth init failed\n");
-               return rc;
-       }
 
        bt_conn_cb_register(&bt_conn_cb);
```

**To Reproduce**
Steps to reproduce the behavior:

```
mkdir build; cd build
export CONF_FILE=prj_96b_nitrogen.conf
cmake -GNinja -DBOARD=nrf52840_pca10056 ..
ninja flash
```

Output:

```
[00:00:00.008,392] <inf> bt_hci_core.hci_vs_init: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.008,392] <inf> bt_hci_core.hci_vs_init: HW Variant: nRF52x (0x0002)
[00:00:00.008,422] <inf> bt_hci_core.hci_vs_init: Firmware: Standard Bluetooth controller (0x00) Version 1.13 Build 99
[00:00:00.008,789] <inf> bt_hci_core.bt_dev_show_info: Identity: c7:fe:f4:25:17:1e (random)
[00:00:00.008,819] <inf> bt_hci_core.bt_dev_show_info: HCI: version 5.0 (0x09) revision 0x0000, manufacturer 0x05f1
[00:00:00.008,819] <inf> bt_hci_core.bt_dev_show_info: LMP: version 5.0 (0x09) subver 0xffff
[00:00:00.008,819] <wrn> bt_hci_core.bt_pub_key_gen: ECC HCI commands not available
[00:00:00.389,648] <err> net_if.join_mcast_allnodes: Cannot join all nodes address ff02::1 (-5)
uart:~$ 
waiting for bt connection: prio recv thread stack (real size 448):	unused 320	usage 128 / 448 (28 %)
recv thread stack (real size 1024):	unused 696	usage 328 / 1024 (32 %)
/
[publisher:375] network_setup: 0 <OK>
prio recv thread stack (real size 448):	unused 188	usage 260 / 448 (58 %)
recv thread stack (real size 1024):	unused 96	usage 928 / 1024 (90 %)
```

(before earlier fix, it would simply exit `network_setup()` and not attempt to connect)

**Expected behavior**

Application should attempt to connect to MQTT server. I do not see any connection attempts. Through the shell, I can actually connect to mosquitto:

```
uart:~$ net tcp connect 2001:db8::2 1883
Connecting from [2001:db8::c7fe:f4ff:fe25:171e]:0 to [2001:db8::2]:1883
uart:~$ prio recv thread stack (real size 448):	unused 152	usage 296 / 448 (66 %)
recv thread stack (real size 1024):	unused 96	usage 928 / 1024 (90 %)
TCP connected
```

```
Mosquitto log:
1543505073: New connection from 2001:db8::c7fe:f4ff:fe25:171e on port 1883.
```

**Impact**
Unable to test a very common use case for BLE/6lowpan, and also things to test after this like TLS support for MQTT.

**Environment (please complete the following information):**
 - OS: Linux raspbian 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
 - Toolchain: zephyr SDK toolchain
 - Tested on master: 8dc8e4d915 (after the above change)

ntavish, 2018-11-29:
CC @rlubos

mike-scott, 2018-11-29:
The various prj*.conf files look quite out-dated and probably need to be cleaned up a bit.

The new standard is to design a nice generic prj.conf file which gets used by default and then overlay-*.conf for various specialized cases.

pfalcon, 2018-11-29:
@ntavish: Thanks for submitting this ticket, was looking to create one myself ;-).

So, @rlubos, the problem here is that this sample should not have stuff like `#if defined(CONFIG_NET_L2_BT)`. We have net_config subsys to handle networking setup.

Also from @mike-scott on IRC:

>  The multiple prj_*.conf files threw me off with all of their net_buf/net_pkt settings
<hashcode> TBH, most of these settings could be cleaned up

rlubos, 2018-11-30:
The fact is BLE stuff in the sample might be outdated, I did not touch it at all during the MQTT update. Will look into this.

pfalcon, 2018-11-30:
> The fact is BLE stuff in the sample might be outdated,

It *is* outdated, as mentioned, stuff like that was replaced long ago by net_app_config which was replaced by net_config. It seems to have some display hacks, like rotating progress indicator waiting for BLE conn, so will be hard to remove ;-).

ntavish, 2018-11-30:
Take a look at this #11765, it doesn't handle connecting properly is BT is not connected, but it does connect if you are fast enough in connecting after boot.

rlubos, 2018-12-03:
Closing as this got fixed by #11765.

closed 2018-12-03 by rlubos