zephyrproject-rtos/zephyr #11743: logging: add user mode access

andrewboie, 2018-11-29:
We'd like to allow user threads to use the logging subsystem, at the moment only supervisor threads may log events.

How to do this will require some thought. We want to minimize latency and also ensure that garbage passed in will not present any opportunities to crash or compromise the kernel. We'd also like to use the same logging APIs in user mode, and not have a parallel set.

Further comments for analysis and brainstorming, I don't have a concrete implementation proposal yet.

pfalcon, 2018-11-30:
> We want to minimize latency

Minimizing latency is always good, but I guess for userspace it's not the same stringent constraints as for kernelspace (where @nordic-krch quotes 250 cycles to log a message, etc.), where number of tricks and optimizations are used to achieve that. For userspace (which is "slow" anyway) hopefully the main requirement is that it just works and easy to maintain, i.e. no those macro meta-programming and stuff.

pizi-nordic, 2018-11-30:
I thought about a bit different approach:

1. Keep the logger core in kernel as is.
2. Add "Userspace" backend, which will render and forward rendered messages to userspace.
3. Create userspace version of the logger. IMHO this should be possible by abstracting logger data structures.
4. Use dedicated frontend to communicate with kernel logging.

So data flow from userspace will look like this:
LOG_MACRO(...) => Logger Core (Userspace) => Backend.

And from kernel like this:
LOG_MACRO(...) => Logger Core (Kernel) => Userspace Backend (Kernel side) => Userspace Frontend (Userspace) => Logger Core (Userspace) => Backend.

What do you think?

andrewboie, 2018-11-30:
> I thought about a bit different approach:
> 
> Keep the logger core in kernel as is.
> Add "Userspace" backend, which will render and forward rendered messages to userspace.
> Create userspace version of the logger. IMHO this should be possible by abstracting logger data structures.
> Use dedicated frontend to communicate with kernel logging.

Yeah I think this makes the most sense.

andrewboie, 2018-11-30:
@pizi-nordic so the more I think about it, I'm pretty convinced that rendering the log strings will need to be done at the log call site, due to the potential for bugs or abuse with "%s".

- The calling thread may not have access to the memory pointed to by the %s pointer. If a log thread does the rendering, the memory will be accessed based on the log thread's memory domain configuration and not the caller's. Could possibly expose information the caller should not have had access to.
- The calling thread can crash the log thread by passing in a completely bad string pointer.

```
And from kernel like this:
LOG_MACRO(...) => Logger Core (Kernel) => Userspace Backend (Kernel side) => Userspace Frontend (Userspace) => Logger Core (Userspace) => Backend.
```

Traversing the user/kernel boundary typically involves data copies. This seems like it might have a lot of overhead.

I'd suggest something a bit simpler:

From supervisor thread - Leave log control flow as-is
From user thread - Render the formatted message at the call site, or at least enough that any printf-like processing is done at the call site. (stuff like rendering timestams could still be deferred) Pass the rendered string to the kernel through a system call, which in turn logs the message as if the kernel did it.

This way we don't need any extra logging threads or backends on the userspace side, we just do some rendering first and pass the rest to the kernel to log as normal.

andrewboie, 2018-11-30:
> Minimizing latency is always good, but I guess for userspace it's not the same stringent constraints as for kernelspace (where @nordic-krch quotes 250 cycles to log a message, etc.), where number of [...]

No, our customers have pretty strict latency requirements on the overhead of making system calls, for example. The code paths need to be tightly optimized.

> i.e. no those macro meta-programming and stuff.

What specifically are you talking about? Do you feel we are doing something unnecessarily?

pfalcon, 2018-11-30:
> What specifically are you talking about? Do you feel we are doing something unnecessarily?

No, I just consider logging implementation relatively complex.

Anyway, if there's a requirement for both userspace/kernel split **and** minimal latency/overhead when doing syscalls, well, will keep this fact in mind. Because IMHO the obvious way to get minimum overhead is to not use syscalls at all with all their TOCTOU workaround overheads, etc.

pizi-nordic, 2018-12-05:
Hi @andrewboie. I have talked with @nordic-krch, who is our logger expert, about the changes required in order to support user space. Results of the discussion are below.

- **Module identification and log level control:**
  The logger uses 2 data structures (`log_source_const_data` and `log_source_dynamic_data`) per each registered module in order to provide metadata for the log core as well as control the logging level. Both structures are located into dedicated sections, which might reside in kernel space. In order to log a message, you have to obtain:

  - Log source ID (please look into `log_const_source_id()` how it is generated)
  - Current filtering setting, which have to be read from the dynamic data

  While identification could be done in userspace, the readout of filter settings have to be done in kernel space. This means, that we have to have syscall equivalent of the `LOG_RUNTIME_FILTER()`. It should be easy to implement as userspace can provide address of data structure which have to be read and checking that information is easy on kernel side (look for `__log_dynamic_start` and `__log_dynamic_end` in the code).

- ** Log Rendering **
  I think that the in-place rendering you proposes is the easiest and the safest method to pass the data to the kernel space. You might look into `log_printk()` in order to see how to insert formatted string into logger infrastructure.

andrewboie, 2019-04-11:
Spent some time looking at this today.

First step is to identify the APIs that are intended to be used at the application level. I spoke to @carlescufi and I think we need some re-namespacing; there are lot of APIs, symbols, or macros which start with log_/LOG_ but seem to be intended for internal use by the logging core, or part of an API between the logging core and various log backends, or are never called directly and are used internally by other public-facing log macros. I really feel these should be under a different namespace, some prefixed with Z_, some under an alternate namespace like log_core_ or log_backend_, something like that. **We need to know what APIs are *contracts* between the logging subsystem and applications that use it.**

The tests under tests/subsys/logging/* use a lot of APIs which do not seem to be application facing. They seem to test various parts of the logging subsystem internals, and I don't think having any of them run in user mode would be useful or desirable.

That leaves the sample under subsys/logging/logger, which I think should be adjusted to run entirely in user mode. What I really want is something under tests, using ztest, that *only* uses application-facing APIs, runs in user mode, and reports success or failure for logging various types of messages. However the existing sample is enough to get started.

Using that as a guide, my first-pass analysis on what APIs are app-facing are so far:

- LOG_*
- LOG_INST_*
- LOG_HEXDUMP_*
- LOG_INST_HEXDUMP_*
- log_printk()
- log_strdup()
- log_generic()
- LOG_MODULE_REGISTER()
- LOG_MODULE_DECLARE()
- LOG_LEVEL_SET()
- LOG_INSTANCE_REGISTER()
- LOG_INSTANCE_PTR_INIT()
- LOG_INSTANCE_PTR_DECLARE()

Some user mode considerations, some of which we have already discussed:

- log_strdup() will be a no-op when called from user mode, since the format string will be rendered at the call site and not deferred.
- I think the system call for logging should be somewhere in the neighborhood of __LOG(); basically immediately after filtering takes place in __LOG() and we know we are going to log the message (and equivalent spot in __LOG_HEXDUMP). The message string will be rendered into a C string before the syscall is made. The system call implementation will copy it with log_strdup() and pass that to the logging infrastructure before returning unless synchronous logging is enabled.
- Since the filtering will take place in the user thread's context, all the related data needs to be accessible. The const data is fine, it will be in rodata and user mode is able to read this. The dynamic data on the other hand, needs to be assignable to a memory domain partition otherwise the thread can't read it. I need to study this some more.

andrewboie, 2019-04-11:
@nordic-krch can you confirm the above list of application-facing APIs?

nordic-krch, 2019-04-12:
@andrewboie the structure of headers were following:
-  `log.h` - API to be used by the modules which are using the logger
- `log_ctrl.h` - API to be used to control the logger - trigger processing (if thread is not used), control filtering, enable backends, etc.
- 'log_instance.h' - API to be used when logger is used to filter on module instance level. Actually, now it can be merged with `log.h` because in the beginning `log.h` mustn't be included by headers because it required local log level to be defined before including (so it was not recommended to include that in headers). This requirement is now removed so in near future 'log_instance.h` API can be merged to `log.h`.
- `log_core.h` - internals taken into separate file to keep `log.h` file from macro magic. However, it is kept in `include/logging` because it is used by the `log.h` and contains API which is an indirect public API (macro is public but resolves at compile time to those functions). `log_is_strdup()` `log_free()`, `log_dropped()` are internal functions, others are used in the api context.

I need to focus on #15370. I will try to provide more details later.

andrewboie, 2019-04-12:
>  contains API which is an indirect public API (macro is public but resolves at compile time to those functions). log_is_strdup() log_free(), log_dropped() are internal functions, others are used in the api context.

There's really no such thing as 'indirect public', it's just the macros that are public. These functions need to be renamed.

andrewboie, 2019-06-05:
I had a little free time about a month ago and spent a little time looking at this.

I think I found a good place to place the syscall, I hooked it into __LOG_INTERNAL().
See WIP patch: https://github.com/andrewboie/zephyr/commit/5be5e81bded2517bbe11811dd95f752489827809

However the runtime filtering I didn't come up with a solution, the problem is user mode thread access to the runtime filtering data structures. I suppose we could just place all of them in a dedicated global memory partition, but it makes me uneasy to do this since this could be a way for otherwise sandboxed areas of the system to interfere with each other.

nordic-krch, 2019-06-07:
@andrewboie i'm not that familiar with userspace but what if there were two copies of filter created for each module, one in kernel space and one in userspace? api call to change the filter level (currently mainly used by shell) would modify both and keep them in sync but __LOG_INTERNAL macro would use kernel copy and syscall second copy. Cost: one word in ram per module. Taking into account that userspace would be usually used by bigger chips that's not much.

nashif, 2019-06-07:
Is there a combination of kconfig options that would disable features that are not compatible with userspace and keeps basic logging functional? i.e. no filtering, immediate delivery, etc.?

andrewboie, 2019-06-07:
> @andrewboie i'm not that familiar with userspace but what if there were two copies of filter created for each module, one in kernel space and one in userspace? api call to change the filter level [...]

I don't see how this would work, can you provide more details

> Is there a combination of kconfig options that would disable features that are not compatible with userspace and keeps basic logging functional? i.e. no filtering, immediate delivery, etc.?

The patch I linked above, if cleaned up, would provide logging without any dynamic filtering.

nordic-krch, 2019-06-10:
> I don't see how this would work, can you provide more details

Runtime filters are created in specific memory section. My idea was to have second memory section in userspace memory. Registration macro would create 2 filter variables in two sections. Log in userspace would check filter variable accessible by userspace. That would add message to the buffer. When log is processed filter is checked again. Then filter copy from kernel is checked. API for controlling filtering for given module by changing both variable, keeping them in sync. Userspace has access to userspace filter but cannot do much harm - can increase filter (will result in more log messages being created) or decrease (less messages) but logger will not process (pass to backend) more than set by logger API.

nashif, 2019-06-14:
@nordic-krch @andrewboie can you document the discussed and proposed solution here please?

nordic-krch, 2019-06-17:
Proposed solution for handling runtime filtering assumes that filtering settings can only be changed from kernel space and user space can only read it. Filters will be stored in memory partition (section?) which will be read only for user space.
To be considered is the fact that such read only memory partition may have other use cases than the logger run time filtering. Some of the syscalls could be replaced by providing read only memory for user space.

andrewboie, 2019-06-17:
Some details:

We need to define a special read-only k_mem_partition, and route all the dynamic runtime filtering data into it. We can call it k_user_rodata_partition, and define it in include/app_memory/partitions.h. The definition should include documentation on what situations it is needed for (in this case, logging).

At the moment, K_APPMEM_PARTITION_DEFINE() has K_MEM_PARTITION_P_RW_U_RW hard-coded. With some adjustments, we could add another API which has the user supply the desired access policy, like K_APPMEM_PARTITION_POLICY_DEFINE(name, access) and just

`#define K_APPMEM_PARTITION_DEFINE(name) K_APPMEM_POLICY_DEFINE(name, K_MEM_PARTITION_P_RW_U_RW)`

This new partition would have K_MEM_PARTITION_P_RO_U_RO access, as most (all?) platforms don't support K_MEM_PARTITION_P_RW_U_RO. If the runtime configuration is being changed, we will need to temporarily lock the scheduler and adjust the MMU/MPU to allow writes to this region. We can't completely disable the MPU/MMU, I do not think this is safe, we would not catch stack overflows during interrupts and we would be disabling cache-ability settings which might be important. We do not need to expose system calls to adjust runtime filter settings, for now we can leave supervisor-only.

Finally, we will need to route all the dynamic filtering stuff into the k_user_rodata_partition. This is done with the K_APP_DMEM()/K_APP_BMEM() macros, which internally places the data in a special section. However, the dynamic filtering stuff is already routed to a section so we will have to figure out how to reconcile this.

rljordan, 2019-08-13:
Please update the status of this.  Will it be fixed for 2.0?

andrewboie, 2019-08-13:
If this can wait until 2.1, and we can agree this is an enhancement / new functionality, then I'm willing to do it.

andrewboie, 2019-08-14:
I'm told this has to be considered a bug for 2.0 for now

rljordan, 2019-08-14:
Anas noted that this is a bug.  Is it possible to have it fixed in the 2.0 time frame?  We have internal users that want this to work in User mode.

PR #18731 (2019-08-29): Backport 18575 to v1.14 branch: "logging: Add basic userspace support" 
This PR adds basic userspace support to the logging subsystem.    
With this change, the following API could be called from user mode:   
- LOG_*()                                                             
- LOG_INST_*(),                                                       
- LOG_HEXDUMP_*(),                                                    
- LOG_HEXDUMP_INST_*(),                                               
- LOG_PANIC(), LOG_PROCESS(),                                         
- log_printk(), log_generic(), log_ [... cut]
closed 2019-09-11 by andrewboie