zephyrproject-rtos/zephyr #11744: Regression: nNRF52840 HW dies at 8 minutes 20 seconds in various samples using IPSP

mike-scott, 2018-11-29:
**Describe the bug**
During automated testing of the LwM2M client in master branch, a regression was noted on both reel_board and nrf52840_pca10056.  The board dies after 8 minutes and 20 seconds.   Other samples which aren't public also displayed the same behavior.  One was an MQTT and HTTP application.  All samples tested use IPSP network connection via BLE.

Bisection led to this commit:
baea22407db46e99c22c7b04bf7413880cd111d3 kernel: Always set clock expiry with sync with timeout module

Prior to this commit, the samples runs normally.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/net/lwm2m_client
2. mkdir build; cd build
3. cmake -DBOARD=reel_board -DOVERLAY_CONFIG=overlay-bt.conf ..
4. make
5. make flash
6. board should be connected via IPSP (timeout might happen during init due to using generic IPv6 addresses -- doesn't affect the outcome)
6a. NOTE: if user wants connection to actually work, they may need to alter the configuration of their IPSP gateway to use the 2001:db8:: subnet.
6b. Outcome is the same whether or not IPSP networking actually works
7. wait for 8 minutes and 20 seconds, board will stop responding

**Expected behavior**
Board / sample should run longer than 8 minutes 20 seconds.

**Impact**
Unacceptable board behavior

**Screenshots or console output**
```
[LOG INCOMING]
```

**Environment (please complete the following information):**
 - OS: Linux
 - Distribution: Ubuntu 18.04
 - Toolchain: Zephyr SDK
 - Commit SHA: baea22407db46e99c22c7b04bf7413880cd111d3

carlescufi, 2018-11-29:
Assigning to @pdunaj, @andyross and @andrewboie since this looks like a commit to the kernel code.

mike-scott, 2018-11-29:
Updated w/ full log from nrf52840_pca10056 running lwm2m_client sample from master with no changes.

andyross, 2018-11-29:
Huh... that patch fixed a real bug, so it can't just be reverted.  This is for sure a rollover condition.  This is a 16kHz clock, right?  So a signed rollover is going to happen at 2^23 bits, or just about exactly eight and a half minutes.  I'll take a look, though @pdunaj  is likely to see it first.

Note that I have another rework in progress of this area to prevent the "set timeout on idle" behavior entirely, which is essentially a parallel way to fix the original bug (with rapidly/repeatedly set timeouts bumping the next tick forward).   If we can't find the rollover here, we could presumably revert this one once the other fix is ready.  If, y'know, it doesn't itself introduce more bugs.

carlescufi, 2018-11-29:
@andyross I think this is [a 32KHz clock](http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.nrf52/dita/nrf52/development/nrf52840_pdk/hw_block_diagram.html?cp=2_0_3_7_1)

aurel32, 2018-11-29:
I guess this bug is a duplicate of #11694.

andyross, 2018-11-29:
Hah, even better as it can be an unsigned overflow. 

Oh!  And I see you literally just pushed a patch.  Yeah, that looks like it exactly.

andyross, 2018-11-29:
(Though AFAICT that bug predates @pdunaj 's patch mentioned above.  @mike-scott are you sure about that bisection?)

mike-scott, 2018-11-29:
@andyross Absolutely.  I spent quite some time working through the commits.  And this bug reproduces perfectly every time.

ntavish, 2018-11-30:
I saw this with MQTT sample also. Logging completely stops, but if after it stops, you connect BLE and do IPSP stuff, the networking is still working, can ping to the device, and I am still getting stack init logs like `prio recv thread stack (real size 448): unused 188      usage 260 / 448 (58 %)`.

pdunaj, 2018-11-30:
@mike-scott , @andyross , bug that was fixed by the commit mentioned caused delayed work to be not executed. It could be that in the code there was something that was simply not running and now it is allowed to execute and behavior changes. If it is so it should be relatively simple to track down.

PR #11747 (2018-11-30): drivers: timer: nrf_rtc_timer: fix handling for 24-bit counter
Two subtractions failed to account for the possibility that a calculated
time exceeded the counter resolution, allowing a comparison to
improperly indicate that a minimum delay was satisfied.

Use the subtraction helper to avoid the problem.

(The subtraction in z_clock_set_timeout was the cause of issue #11694;
the one in rtc1_nrf5_isr was replaced based on inspection rather than
testing.)

Closes #11694 
Fixes #11744
closed 2018-11-30 by carlescufi