zephyrproject-rtos/zephyr #11745: logging: never leaves panic mode on fatal thread exception

andrewboie, 2018-11-29:
**Describe the bug**
If a thread crashes, the kernel can decide to hang the system, or just kill the offending thread. The log subsystem enters panic mode to flush the logs and emit the exception data synchronously, but if the thread is simply killed the log subsystem never leaves panic mode.

**Expected behavior**
If the thread is simply killed, an API call is made to leave panic mode immediately before _SysFatalErrorHandler calls k_thread_abort().

**Impact**
If user mode is not used, low, since a fatal thread exception makes no guarantees on data corruption, the thread could have been executing kernel code leaving shared data structures in a bad state anyway, or blew its stack and messed up adjacent data structures. A supervisor mode thread crashing is always very bad news.

If the crashed thread was a user thread, then we can at least assert the core kernel is in a good state and this becomes more of an issue since the logging subsystem overhead increases greatly.

nordic-krch, 2019-01-07:
@andrewboie can you help me with that? Is there a place in code code where we can be sure that system will not recover?

galak, 2019-03-27:
@andrewboie, @nordic-krch any updates on this?

nordic-krch, 2019-04-01:
no update. maybe @andrewboie could help me with answering question above.

andrewboie, 2019-04-04:
> @andrewboie can you help me with that? Is there a place in code code where we can be sure that system will not recover?

Just look at SysFatalErrorHandler, it makes a decision on whether the thread should simply be killed or the whole system taken down.

nashif, 2020-03-11:
@nordic-krch  ping

nordic-krch, 2020-03-12:
error handling has been reworked by @andrewboie and `LOG_PANIC` is now called in `k_sys_fatal_error_handler`. It seems to me that this function is called only on unrecoverable error?@andrewboie can you confirm? If yes then issue can be closed.

andrewboie, 2020-03-12:
> error handling has been reworked by @andrewboie and `LOG_PANIC` is now called in `k_sys_fatal_error_handler`. It seems to me that this function is called only on unrecoverable error?@andrewboie can you confirm? If yes then issue can be closed.

With the invocation of LOG_PANIC() moved to k_sys_fatal_error_handler(), I agree.

closed 2020-03-12 by andrewboie