zephyrproject-rtos/zephyr #11752: tests/lib/ringbuffer: build error on Xtensa ESP32

inakypg, 2018-11-29:
Commit ID 2b1d54e897fed4cf1de7c641408f3d90676c31bc

ESP-IDF https://github.com/espressif/esp-idf.git commit dc8c33892e0 and b2ff235bd00

```
...
[ 65%] Building C object zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/irq_offload.c.obj
In file included from /opt/esp-idf.git/components/soc/esp32/include/soc/dport_reg.h:17:0,
                 from zephyr.git/soc/xtensa/esp32/soc.h:9,
                 from zephyr.git/arch/xtensa/core/irq_offload.c:13:
/opt/esp-idf.git/components/soc/esp32/include/soc/soc.h:128:0: error: "BIT" redefined [-Werror]
 #define BIT(nr)                 (1UL << (nr))
 ^
In file included from zephyr.git/include/sched_priq.h:9:0,
                 from zephyr.git/include/kernel_includes.h:23,
                 from zephyr.git/include/kernel.h:17,
                 from zephyr.git/arch/xtensa/core/irq_offload.c:6:
zephyr.git/include/misc/util.h:135:0: note: this is the location of the previous definition
 #define BIT(n)  (1UL << (n))
 ^
cc1: all warnings being treated as errors
```

Full log

```
$ mkdir -p /tmp/tcf.run-_lRxhf/bhdz/outdir-bhdz-ycbk-esp32
$ rm -f /tmp/tcf.run-_lRxhf/bhdz/outdir-bhdz-ycbk-esp32/.config
$ cmake -DBOARD=esp32 -DARCH=xtensa -DEXTRA_CPPFLAGS="-DTC_RUNID=ci-181129-0928-138:ycbk" -DEXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" -DEXTRA_AFLAGS=-Wa,--fatal-warnings -DEXTRA_LDFLAGS=-Wl,--fatal-warnings  -B"/tmp/tcf.run-_lRxhf/bhdz/outdir-bhdz-ycbk-esp32" -H"zephyr.git/tests/lib/ringbuffer"
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.7", minimum required is "3.4")
-- Selected BOARD esp32
Zephyr version: 1.13.99

warning: ENTROPY_ESP32_RNG (defined at drivers/entropy/Kconfig.esp32:7) was assigned the value 'y'
but got the value 'n'. You can check symbol information (including dependencies) in the 'menuconfig'
interface (see the Application Development Primer section of the manual), or in the Kconfig
reference at http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_ENTROPY_ESP32_RNG.html
(which is updated regularly from the master branch). See the 'Setting configuration values' section
of the Board Porting Guide as well.
Parsing Kconfig tree in zephyr.git/Kconfig
Loading zephyr.git/boards/xtensa/esp32/esp32_defconfig as base
Merging zephyr.git/tests/lib/ringbuffer/prj.conf
Merging /tmp/tcf.run-_lRxhf/bhdz/outdir-bhdz-ycbk-esp32/500_boot_config.conf
-- Cache files will be written to: /home/jenkins/.cache/zephyr
CMake Warning at zephyr.git/boards/common/esp32.board.cmake:5 (message):
  Setting ESP_IDF_PATH in the environment is deprecated.  Use cmake
  -DESP_IDF_PATH=...  instead.
Call Stack (most recent call first):
  zephyr.git/boards/xtensa/esp32/board.cmake:1 (include)
  zephyr.git/cmake/app/boilerplate.cmake:316 (include)
  CMakeLists.txt:2 (include)

-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
CMake Warning at ../../../CMakeLists.txt:1335 (message):

        ------------------------------------------------------------
        --- WARNING:  __ASSERT() statements are globally ENABLED ---
        --- The kernel will run more slowly and use more memory  ---
        ------------------------------------------------------------

-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tcf.run-_lRxhf/bhdz/outdir-bhdz-ycbk-esp32
$ make -C /tmp/tcf.run-_lRxhf/bhdz/outdir-bhdz-ycbk-esp32 config-sanitycheck' @tcf.git/tcfl/app_zephyr.py:310
Scanning dependencies of target config-sanitycheck
[... 75 lines cut ...]
[ 60%] Built target drivers__gpio
[ 61%] Building C object zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/stdout/fprintf.c.obj
[ 62%] Linking C static library liblib__libc__minimal.a
[ 62%] Built target lib__libc__minimal
[ 63%] Building C object zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2.c.obj
[ 64%] Building ASM object zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/atomic.S.obj
[ 65%] Building C object zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/irq_offload.c.obj
In file included from /opt/esp-idf.git/components/soc/esp32/include/soc/dport_reg.h:17:0,
                 from zephyr.git/soc/xtensa/esp32/soc.h:9,
                 from zephyr.git/arch/xtensa/core/irq_offload.c:13:
/opt/esp-idf.git/components/soc/esp32/include/soc/soc.h:128:0: error: "BIT" redefined [-Werror]
 #define BIT(nr)                 (1UL << (nr))
 ^
In file included from zephyr.git/include/sched_priq.h:9:0,
                 from zephyr.git/include/kernel_includes.h:23,
                 from zephyr.git/include/kernel.h:17,
                 from zephyr.git/arch/xtensa/core/irq_offload.c:6:
zephyr.git/include/misc/util.h:135:0: note: this is the location of the previous definition
 #define BIT(n)  (1UL << (n))
 ^
cc1: all warnings being treated as errors
make[3]: *** [zephyr/CMakeFiles/zephyr.dir/build.make:143: zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/irq_offload.c.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:537: zephyr/CMakeFiles/zephyr.dir/all] Error 2
make[1]: *** [Makefile:84: all] Error 2
```

spoorthik, 2018-12-11:
The issue #11077 complains the same.

nashif, 2019-01-11:
duplicate of #11077

closed 2019-01-11 by nashif