zephyrproject-rtos/zephyr #11763: logger drops log messages without any indication

nordic-krch, 2018-11-30:
**Describe the bug**
When logger is extensively used it may happen that new log message cannot be allocated. There are two options for the logger to react `LOG_MODE_OVERFLOW` or `LOG_MODE_NO_OVERFLOW`. If overflow is enabled then oldest logs are dropped. There is no indication of that so user may get confused by the log content and do not know that internal logger buffer should be increased

**To Reproduce**
Decrease `LOG_BUFFER_SIZE` or add more logs to the sample and see that log output does not contain all logs.

**Expected behavior**
User should be warned about that situation.

**Impact**
Logs confuses the user.

nordic-krch, 2018-11-30:
Originates from #11655

nordic-krch, 2018-11-30:
fyi @pfalcon @andrewboie @nashif

pfalcon, 2018-11-30:
> Expected behavior
User should be warned about that situation.

As argued in https://github.com/zephyrproject-rtos/zephyr/issues/11655, there should be 2 choices:

1. warn user about about the situation (i.e. produce a suitable log message)
2. switch to truly sync mode in this case

I'm fine if this ticket is dedicated to just p.1, p.2 is then tracked in #11655.

nordic-krch, 2018-11-30:
There are two options: notification vs polling.
- add `log_drop_count_get()` to logger api. Backends asks about current drop and reacts (e.g. reports every log losted, or asks periodcally). It's up to backend what to do with that information.
- extend backend API with log_dropped() call. Similar thing. Backend gets notified and reaction is up to it.

I'm leaning toward notification as polling would need to be performed always, even when buffer size is sufficient (ideally that's close to always if user reacts to 'log dropped' warning.

pfalcon, 2018-11-30:
I'm sure people with more knowledge of logging impl will comment. I know little about it so far, and all the above sounds complex to me. 

My thinking would be along the lines of: Is logging buffer global or per-backend? If per-backend, d'oh, need to duplicate solution for each. Otherwise, when adding a new entry, see if too little space is left, and if so, instead of adding the requested entry, log message "out of log buffer space, some message are lost, increase CONFIG_FOO".

The issue is that such message can be lost too under some circumstances, leaving a user again in the cold. That's why p.2 solution above is important (and would be preferred by me).

If all what you describe is exactly to avoid this "message about lost data is itself lost" case, then sure, good thinking.

nordic-krch, 2018-11-30:
The message pool is common for the logger, then log message is passed to all backends (after filtering if runtime filtering is enable). Reference counter in the message makes sure that when last backend releases the message it is returned to the pool.

yes, i want to avoid allocating 'dropped log' message as it can get lost as well. Dedicated indication is lightweight and leaves to the backend how to react: it can be `~` printed for each log message instead of printing long, verbal warning. Also by seeing number of dropped message it can give user rough estimation how much buffer must be increased.

pfalcon, 2018-12-06:
@nordic-krch: Now that https://github.com/zephyrproject-rtos/zephyr/pull/11769 was merged, a friendly ping about next steps.

jhedberg, 2018-12-12:
Also see #12040 which I just opened about dropped messages even though there should be more than enough buffers available.

nordic-krch, 2019-01-30:
#11769 (generic mechanism), #12602 (extention to uart,rtt, native posix backends), #12664 (shell backend extended) all together fixed the issue.

closed 2019-01-30 by nordic-krch