zephyrproject-rtos/zephyr #11767: tests: counter api: Failed when MPU enabled

erwango, 2018-11-30:
**Describe the bug**
in https://github.com/zephyrproject-rtos/zephyr/pull/11729, 
tests/drivers/counter/counter_basic_api throws a MPU Fault when MPU is enabled.

**To Reproduce**
Steps to reproduce the behavior:
1. CONFIG_ARM_MPU=y
1. mkdir build; cd build
2. cmake -DBOARD=nucleo_f746zg
3. ninja flash
4. See error

**Expected behavior**
Test runs as when MPU is disabled

**Impact**
Can't test counter api when MPU is enabled.

**Screenshots or console output**
```
***** Booting Zephyr OS zephyr-v1.13.0-1890-g973796a *****
***** MPU FAULT *****
  Data Access Violation
  MMFAR Address: 0x2000280c
**** Unknown Fatal Error 2! ****
Current thread ID = 0x2000120c
Faulting instruction address = 0x800ab22
Fatal fault in essential thread! Spinning...

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA : https://github.com/zephyrproject-rtos/zephyr/pull/11729

**Additional context**
Same issue on nucleo_f401re, disco_l475_iot1

erwango, 2018-12-03:
Coming from stack guard misconfiguration (or at least conflict with application):

0 rasr=8000000 rbar=7020025 P_RO_U_RO(8000000 8080000 , subreg 0)
1 rasr=20000001 rbar=110b001f X P_RW_U_NA(20000000 20010000 , subreg 0)
2 rasr=20010002 rbar=110b001d X P_RW_U_NA(20010000 20018000 , subreg 0)
3 rasr=20000003 rbar=130b0017 X P_RW_U_RW(20000000 20001000 , subreg 0)
4 rasr=20002804 rbar=150b0009 X P_RO_U_NA(20002800 20002820 , subreg 0)

We're attempting to write into 2000280c

erwango, 2018-12-03:
Issue is solved by increasing CONFIG_MAIN_STACK_SIZE from 512 to 1024.
It appears this is default value and it is reduced down to 512 if ZTEST.
This has been introduced with bd9a1548ace9e717d5103e21f7a7e02776b246f1

erwango, 2018-12-03:
@galak , @nashif .

I'm questioning bd9a1548ace9e717d5103e21f7a7e02776b246f1 which prevents some tests to run on STM32 targets. I've tried changing to 768, but overflow still happens (bit later).
As commit message says, I've tried to increase (doubled) CONFIG_ZTEST_STACKSIZE, but issue still present.
What would the impacts of reverting fully the change?

andrewboie, 2018-12-03:
What's special about this test that only it is failing?
If it's using ztest, is it somehow doing a lot of processing in the main thread?

galak, 2019-01-15:
Can you move this comments onto the PR #12475

erwango, 2019-01-16:
@galak, PR #12475 has been merged. Should I create a specific PR?

galak, 2019-02-08:
Is this still the case w/new counter API & driver?

erwango, 2019-02-08:
No, I've been able to run tests w/o. Closing. Will reopen if needed

closed 2019-02-08 by erwango