zephyrproject-rtos/zephyr #11770: Multiprotocol feature for BLE/Thread

auschel, 2018-11-30:
Hello

I'm currently working on a project which should achieve the following: A node works as a low power Thread end device, while also be able to establish a BLE connection with a smartphone for configuration purposes. The idea is that Thread and BLE use the same radio concurrently. As far as i've seen, i have to completly deinitialize one stack and initializing the other in order to achive a common usage of the radio because the Thread and BLE stack each have a radio driver of their own. I would like bypass the constant deinitializing & initializing and instead regulate the access to the radio, giving each of the stacks a time window to execute their tasks.

I've seen that Nordic provides multiprotocol support and examples in their SDK and that they are also a member of Zephyr. My use case would be the Dynamic Multiprotocol for Thread and BLE. Is there a way to implement a similar Dynamic Multiprotocol in Zephyr?

markus-becker-tridonic-com, 2020-09-21:
@auschel does https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_multiprotocol_support.html meet your needs?

markus-becker-tridonic-com, 2020-11-23:
@carlescufi I believe this issue could be closed with the information about MPSL in the Nordic Connect SDK.

carlescufi, 2020-12-07:
> @carlescufi I believe this issue could be closed with the information about MPSL in the Nordic Connect SDK.

Agreed @markus-becker-tridonic-com 

Please refer to nRF Connect SDK for this fuctionality:
https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_multiprotocol_support.html

closed 2020-12-07 by carlescufi