zephyrproject-rtos/zephyr #11780: Bluetooth: Mesh: initialisation delay after disabling CONFIG_BT_DEBUG_LOG

vikrant8051, 2018-11-30:
It is found that we have to have enable CONFIG_BT_DEBUG_LOG=y
then only Mesh initialization happens within second otherwise it takes around 10+ seconds. This bug has introduced after latest new logging subsystem.

**To Reproduce**
1) flash samples/boards/nrf52/mesh/onoff_level_lighting_vnd_app in nRF52840_pca10056 
   with CONFIG_BT_DEBUG_LOG=n (it is by default disable as per current prj.conf)
2) Provision & configure it using nRFMesh App
3) Reset the Board & see device serial terminal

Here NODE takes 10+ seconds to print "Mesh initialized"

jhedberg, 2018-12-04:
Looks like this is caused by commit 4d94257162b22f37104e9f85238ed7c3486c5a1c. @vikrant8051 could you try reverting that (`git revert 4d94257162b22f37104e9f85238ed7c3486c5a1c`) and see if the issue goes away for you?

vikrant8051, 2018-12-04:
OK

vikrant8051, 2018-12-04:
@jhedberg 
I think 
> git revert 4d94257162b22f37104e9f85238ed7c3486c5a1c

has no effect....still Mesh initialization takes some time.

jhedberg, 2018-12-04:
@vikrant8051 you also need #11842 . Both PRs should hopefully be merged soon.

jhedberg, 2018-12-04:
@vikrant8051 both PRs are in master now, so you can just rebase on top of that to see if the issue is now solved.

closed 2018-12-04 by jhedberg

vikrant8051, 2018-12-04:
@nordic-krch  @jhedberg @carlescufi 
Issue has resolved !!
Thanks for solving it.

PR #11842 (2018-12-04): drivers: clock_control: Enable LFSTARTED event for nrf5
LFSTARTED interrupt was disabled so it was impossible to put CPU to
sleep and wake up on LF clock being ready. Bluetooth stack was
putting CPU to sleep expecting to be waken up on LFSTARTED event.
As it never occured CPU was waken up seconds later by different event
introducing mulitsecond startup delay.

Fixes #11780.