zephyrproject-rtos/zephyr #11789: Can zephyr BLE mesh support up to 10KB/s p2p throughput?

gofortime, 2018-12-01:
Hi,

Recently I am investigating on adopting a possible BLE Mesh solution in my project. Just found there is BLE mesh support in zephyr now. I am wondering how many nodes can zephyr mesh support? And how about the throughout in point-to-point transfer (e.g. can it support up to 10KB/s throughput in BLE mesh OTA)?

Thanks in advance for any answer to this question. It'd be better if any document or link.

Thanks,
Vincent.

jhedberg, 2018-12-01:
The specifications are available here:

https://www.bluetooth.com/specifications/mesh-specifications

gofortime, 2018-12-02:
> As for your throughput question, section 2.3.9.4 in the Mesh Profile Specification states "a node should not send more than 100 Network PDUs in any 10 second window", which translates to 10 Network PDUs per second on average. An unsegmented message has a 15 byte payload, out of which 4 bytes are reserved for the MIC and 1-3 bytes for the OpCode, so you'll get 8-10 bytes of useful payload in practice. Combine that with 10 PDUs per second, and you get a throughput of max 100 bytes per second, which is far from your 10KB/s target. Note that this is for Mesh version 1.0 - future versions may improve on this using e.g. extended advertising.

Hi Johan, thanks a lot for the detailed information. This is very helpful for my evaluation.

carlescufi, 2018-12-03:
Question answered.

closed 2018-12-03 by carlescufi