zephyrproject-rtos/zephyr #11792: drivers: nrf: UARTE interrupt driven cannot be compiled

jarz-nordic, 2018-12-02:
Interrupt driven API for UARTE driver is causing compilation errors.

**To Reproduce**
Try to build hello world sample for nrf52840 with interrupts support and driver UARTE.

**Expected behavior**
Application compiles and works.

**Impact**
showstopper

jarz-nordic, 2018-12-03:
I think that UART/UARTE selection shall be only in one place: DT to avoid such errors.

Mierunski, 2018-12-03:
To use either UART or UARTE proper compatible has to be selected in dts. As uart0 in nrf52840 can be both nrf-uart and nrf-uarte so to change it, proper overlay file should be added and desired driver should be selected in menuconfig. This is not a bug, as UARTE can be used.

However better way to do it would be possibility to have more than 1 compatible for node.

closed 2018-12-03 by Mierunski