zephyrproject-rtos/zephyr #11794: cmake: CMake 3.13 doesn't like INTERFACE with target_link_libraries

yashi, 2018-12-02:
**Describe the bug**
CMake 3.13 start complaining about CMP0079: `target_link_libraries()` allows use with targets in other directories.

**To Reproduce**
Just make for some board.

**Expected behavior**
No warning while build

**Impact**
It builds fine but prints the warning a few times.

**Screenshots or console output**
```
CMake Warning (dev) at zephyr/cmake/extensions.cmake:837 (target_link_libraries):
  Policy CMP0079 is not set: target_link_libraries allows use with targets in
  other directories.  Run "cmake --help-policy CMP0079" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  Target

    zephyr_interface

  is not created in this directory.  For compatibility with older versions of
  CMake, link library

    -ldl

  will be looked up in the directory in which the target was created rather
  than in this calling directory.
Call Stack (most recent call first):
  zephyr/cmake/extensions.cmake:1055 (target_link_libraries_ifdef)
  zephyr/cmake/extensions.cmake:115 (target_ld_options)
  zephyr/arch/posix/CMakeLists.txt:30 (zephyr_ld_options)
This warning is for project developers.  Use -Wno-dev to suppress it.
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: CMake 3.13.1
 - Commit SHA: d38aa1f

SebastianBoe, 2018-12-03:
I believe we want to short-term use the cmake_policy() command to set this policy to OLD, and then later on, when necessary, port the code-base to the NEW behaviour and set the policyt to NEW.

tejlmand, 2018-12-05:
> I believe we want to short-term use the cmake_policy() command to set this policy to OLD, and then later on, when necessary, port the code-base to the NEW behavior and set the policy to NEW.

basically there is no porting needed if allowing the new behavior, current code works on `CMP0079 NEW`. The reason to state `CMP0079 OLD` is in order to still allow users to use CMake versions 3.8.2 - 3.13

PR #11875 (2018-12-07): cmake: Fix warning related to policy CMP0079
In CMake 3.13 a warning is produced if one does not explicitly state
what the policy for CMP0079 should be. To resolve this warning we set
policy CMP0079 to OLD.

This is expected to have no semantical change. When we want a
semantical change we can flip the policy and port the build scripts to
the new behaviour.

This fixes #11794
commit dc04f552 (2018-12-07): cmake: Fix warning related to policy CMP0079

In CMake 3.13 a warning is produced if one does not explicitly state
what the policy for CMP0079 should be. To resolve this warning we set
policy CMP0079 to OLD.

This is expected to have no semantical change. When we want a
semantical change we can flip the policy and port the build scripts to
the new behaviour.

This fixes #11794
closed 2018-12-07 by nashif