zephyrproject-rtos/zephyr #11803: USB conformance tests

carlescufi, 2018-12-03:
There are 2 sets of USB conformance tests that the Zephyr USB stack should be compliant with:

* [USB3CV tool](https://www.usb.org/usb32tools). This is part of the official USB-IF set of tools
* [Linux usb-test](http://www.linux-usb.org/usbtest/). This is a set of tests written to verify the conformance to the USB spec against the Linux kernel

This issue tracks the status of the different tests against the Zephyr USB stack:

### USB3CV

### Linux usb-test

carlescufi, 2018-12-03:
>The tests are never just against the stack, respective controller driver or shim driver is also heavily involved, especially on STM32 and nRF52.

@jfischer-phytec-iot yes, that's why on the table there's a column for the respective drivers

pawelzadrozniak, 2018-12-05:
I have noticed an issue with testusb aplication build from kernel 4.15.0 - test case 14 and 21 always fail (invalid argument). Works fine on 4.13.0. Have anyone else tried 4.15?

jfischer-phytec-iot, 2018-12-12:
Tested `usb_dc_kinetis.c` with testusb tool, zephyr master eb9f13e3e2f8a97bbfaa0ae83705d610457b9251, host 4.18.0-2-amd64

nzmichaelh, 2020-07-01:
Once #26288 lands, testusb also passes on the SAM0 based Arduino Zero.  I don't have Windows so I can't run USBCV3.

Note the "-v 512" to limit the vary.  Without this you hit a bug in the test driver itself.

```
 $ sudo ~/src/linux-source-5.4.0/tools/usb/testusb -D /dev/bus/usb/003/023 -v 512
/home/michaelh/src/linux-source-5.4.0/tools/usb/testusb: /dev/bus/usb/003/023 may see only control tests
/dev/bus/usb/003/023 test 0,    0.000007 secs
/dev/bus/usb/003/023 test 1,    2.122639 secs
/dev/bus/usb/003/023 test 2,    1.976459 secs
/dev/bus/usb/003/023 test 3,    1.126099 secs
/dev/bus/usb/003/023 test 4,    0.978835 secs
/dev/bus/usb/003/023 test 5,   50.055506 secs
/dev/bus/usb/003/023 test 6,   56.671430 secs
/dev/bus/usb/003/023 test 7,   30.818122 secs
/dev/bus/usb/003/023 test 8,   29.223321 secs
/dev/bus/usb/003/023 test 9,    0.993517 secs
/dev/bus/usb/003/023 test 10,    4.000794 secs
/dev/bus/usb/003/023 test 11,   16.533300 secs
/dev/bus/usb/003/023 test 12,   17.067570 secs
/dev/bus/usb/003/023 test 13,    6.192024 secs
/dev/bus/usb/003/023 test 14,    0.857532 secs
/dev/bus/usb/003/023 test 17,    2.076458 secs
/dev/bus/usb/003/023 test 18,    1.810554 secs
/dev/bus/usb/003/023 test 19,    2.057047 secs
/dev/bus/usb/003/023 test 20,    1.961440 secs
/dev/bus/usb/003/023 test 21,    1.084660 secs
/dev/bus/usb/003/023 test 24,   60.971644 secs
/dev/bus/usb/003/023 test 27,   59.683283 secs
/dev/bus/usb/003/023 test 28,   56.814628 secs
/dev/bus/usb/003/023 test 29,    0.804098 secs
```

nzmichaelh, 2020-07-06:
The SAM0 fixes have landed.

nashif, 2022-11-20:
now sure what part of this is an enhancement, if we are to track usb test results, we need to do this somewhere else, coverting this to Meta now.

jfischer-no, 2025-07-02:
This issue is about the legacy USB device support and is outdated. If there is any demand to track it, a new issue will be opened.