zephyrproject-rtos/zephyr #11827: gPTP Sample Application fails on frdm_k64f board ,and PDelay Response Receipt Timeout

wanhao2015, 2018-12-04:
I'm trying to run gptp demo of zephyr in the frdm_k64f board, （https://docs.zephyrproject.org/latest/samples/net/gptp/README.html）
and run gptp on ubuntu 16.04 with i210 interface card(not using a VM).（https://github.com/AVnu/gptp）
They were wired together.But the question occured,I try to execute the following order in the ubuntu 16.04:
sudo ./gptp enp5s0 -F ../gptp_cfg.ini
INFO     : GPTP [11:09:48:022] gPTP starting
INFO     : GPTP [11:09:48:022] priority1 = 248
INFO     : GPTP [11:09:48:022] announceReceiptTimeout: 3
INFO     : GPTP [11:09:48:022] syncReceiptTimeout: 3
INFO     : GPTP [11:09:48:022] LINKSPEED_100MB - PHY delay
			TX: 1044 | RX: 2133
INFO     : GPTP [11:09:48:022] LINKSPEED_1G - PHY delay
			TX: 184 | RX: 382
INFO     : GPTP [11:09:48:022] neighborPropDelayThresh: 800
INFO     : GPTP [11:09:48:022] syncReceiptThreshold: 8
ERROR    : GPTP [11:09:48:023] Using clock device: /dev/ptp1
STATUS   : GPTP [11:09:48:023] Starting PDelay
STATUS   : GPTP [11:09:48:023] Link Speed: 100000 kb/sec
STATUS   : GPTP [11:09:51:023] *** Announce Timeout Expired - Becoming Master
STATUS   : GPTP [11:09:51:023] New Grandmaster "2C:4D:54:FF:FE:44:9B:9A" (previous "00:00:00:00:00:00:00:00")
EXCEPTION: GPTP [11:09:51:056] PDelay Response Receipt Timeout
EXCEPTION: GPTP [11:09:52:064] PDelay Response Receipt Timeout
EXCEPTION: GPTP [11:09:53:068] PDelay Response Receipt Timeout
EXCEPTION: GPTP [11:09:54:072] PDelay Response Receipt Timeout
EXCEPTION: GPTP [11:09:55:076] PDelay Response Receipt Timeout
EXCEPTION: GPTP [11:09:56:080] PDelay Response Receipt Timeout
EXCEPTION: GPTP [11:09:57:084] PDelay Response Receipt Timeout

How can I solve this problem?Any help will be highly appreciated ! Thank you very much!!!

jukkar, 2018-12-05:
The fix in #11878 fixes the issue where zephyr gptp was dropping all the incoming packets. That issue is fixed, unfortunately it did not fix the "PDelay Response Receipt Timeout" issue that OpenAvnu is reporting. So a bit more TLC is needed here.

jukkar, 2018-12-07:
Various generic gptp issues are now fixed.
So I no longer see these messages
```
EXCEPTION: GPTP [11:09:51:056] PDelay Response Receipt Timeout
```
but there is still some fishy going on in frdm as now I am seeing these kind of messages
```
ERROR : GPTP [16:45:09:99] Link delay -194555505184606449 beyond neighborPropDelayThresh; not AsCapable
```

closed 2018-12-07 by jukkar

jukkar, 2018-12-10:
I am currently investigating why this error started to happen, I do not yet have a solution for this issue.

jukkar, 2018-12-10:
#11991 fixes the negative link delay issue in frdm_k64f

wanhao2015, 2018-12-11:
Thank you for help! I updated the code,and ran gptp again, it succeed!
`
STATUS   : GPTP [11:29:28:768] AsCapable: Enabled
`
Thanks a lot!!!

commit ff0483c5 (2018-12-07): net: gptp: Get the gPTP header properly

If the link layer header is in the separate net_buf,
then skip that one.

Fixes #11827
commit 7503228a (2018-12-11): drivers: eth: mcux: Fix gPTP header parsing

We were reading gPTP header from wrong position when parsing
it in RX and TX.

Fixes #11827