zephyrproject-rtos/zephyr #11841: Assert invoked in exception in ctrl.c line 7653 (zephyr 1.13.00)

alexsven, 2018-12-04:
Assert gets invoked when having one connection and starting another. Spoke to @cvinayak about this and it seems the new connection wants to resize the rx_pool while it's still in use by the first connection. 
`if (_radio.packet_rx_data_count != free_count_rx) {
  /** TODO another role instance has obtained
    * memory from rx pool.
    */
    //LL_ASSERT(0);
} `

carlescufi, 2018-12-05:
@cvinayak can you comment on this issue?

cvinayak, 2018-12-18:
@alexanderwachter @carlescufi this is a bug and needs fix (redesign, as I dont like the current implementation anyway) which I will work alongwith the design to be put in place for the split architecture work.

carlescufi, 2018-12-18:
@cvinayak so this will be fixed in the new split controller only?

cvinayak, 2018-12-20:
@alexanderwachter commenting the assert has unblocked your usecase right? I think subsequent code must be handling/deferring the data resize gracefully, right?

alexanderwachter, 2018-12-21:
I think you guys have a little typo with the names. @alexanderwachter should be @alexsven ;-)

alexsven, 2019-01-02:
@cvinayak Yes, it unblocked our usecase.

carlescufi, 2019-03-05:
@cvinayak can you update on this one please?

cvinayak, 2019-03-15:
@alexsven could you verify #14553 ?

PR #14553 (2019-03-15): Bluetooth: controller: Workaround multi-link DLE assert
Workaround from asserting when Rx PDU buffers from pool
needing resize is acquired by another connection. By
skipping the connection events to give some headroom for
the host to process the Rx packets and eventually make the
pool available for resize.

Fixes #11841.
commit 0437fd7a (2019-03-15): Bluetooth: controller: Workaround multi-link DLE assert

Workaround from asserting when Rx PDU buffers from pool
needing resize is acquired by another connection. By
skipping the connection events to give some headroom for
the host to process the Rx packets and eventually make the
pool available for resize.

Fixes #11841.
closed 2019-03-15 by carlescufi