zephyrproject-rtos/zephyr #11844: rtc_ll_stm32.c:232: undefined reference to `LL_RCC_LSE_SetDriveCapability'  #stm32 #rtc

StefJar, 2018-12-04:
I am having a custom based board on base of the nucleo_f412zg. Activating the RTC at LSE work fine through compilation, but breaks the linking.

Error message:

at linking libdrivers__serial.a:
```
Linking C static library libdrivers__serial.a
/home/stefan/zephyr/drivers/rtc/rtc_ll_stm32.c: In function ‘rtc_stm32_init’:
/home/stefan/zephyr/drivers/rtc/rtc_ll_stm32.c:232:2: warning: implicit declaration of function ‘LL_RCC_LSE_SetDriveCapability’ [-Wimplicit-function-declaration]
  LL_RCC_LSE_SetDriveCapability(CONFIG_RTC_STM32_LSE_DRIVE_STRENGTH);

```
and final at:

```
Linking C executable zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       93976 B         1 MB      8.96%
            SRAM:       29980 B       256 KB     11.44%
        IDT_LIST:         296 B         2 KB     14.45%
drivers/rtc/libdrivers__rtc.a(rtc_ll_stm32.c.obj): In function `rtc_stm32_init':
/home/stefan/zephyr/drivers/rtc/rtc_ll_stm32.c:232: undefined reference to `LL_RCC_LSE_SetDriveCapability'
collect2: error: ld returned 1 exit status

```

Thinks something like:

```
config USE_STM32_HAL_RCC
	default y

config USE_STM32_HAL_RCC_EX
	default y
```
needs to be added

erwango, 2018-12-05:
@StefJar, actually LSE drive strenght is not available on STM32F4.
So, this line should not be used in F4 devices.

STM32 RTC is going to be deprecated soon dur to RTC API depreaction.
It is being replaced by https://github.com/zephyrproject-rtos/zephyr/pull/11729.
So, I'll fix issue directly in #11729 and close this issue.

closed 2018-12-05 by erwango

StefJar, 2019-03-29:
Hey ho
I am try to build a own test project on the latest commit (d3bb3cfd7a4a2358ba7927d5913ecbeb5cfaf577) with RTC for a stm32f412 based custom board. It fails with:

```
Memory region         Used Size  Region Size  %age Used
           FLASH:      104484 B         1 MB      9.96%
            SRAM:       40572 B       256 KB     15.48%
        IDT_LIST:         312 B         2 KB     15.23%
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: drivers/rtc/libdrivers__rtc.a(rtc_ll_stm32.c.obj): in function `rtc_stm32_init':
/home/stefan/zephyr/drivers/rtc/rtc_ll_stm32.c:235: undefined reference to `LL_RCC_LSE_SetDriveCapability'
```
@ @erwango looks the power driver issue is not fixed. Can you please have a look.

a quick setup:
```
CONFIG_ARM=y
CONFIG_SOC_SERIES_STM32F4X=y
CONFIG_SOC_STM32F412ZG=y
CONFIG_RTC=y
CONFIG_RTC_STM32=y
CONFIG_RTC_STM32_CLOCK_LSE=y
# the stm32f4 don't support lse driving
CONFIG_RTC_STM32_LSE_DRIVE_LOW=y
```
thx

erwango, 2019-03-29:
@StefJar , RTC driver is to be deprecated. It should now be replaced by drivers/counter/counter_ll_stm32_rtc.c. Issue is fixed for this driver.
So we can fix the issue to avoid compilation issue in rtc_ll_stm32.c, but for your development, I'd advise to move to counter RTC usage as a more future proof solution.