zephyrproject-rtos/zephyr #11857: cmake does not detect most recent python, it is fixed somehow to 3.4.x

nashif, 2018-12-05:
**Describe the bug**
On a system with multiple versions of python where the binary **python** actually points to python 3.7, cmake still thinks I want 3.4.x.

This is hardcoded in the boilerplate, when I change:

```diff
diff --git a/cmake/app/boilerplate.cmake b/cmake/app/boilerplate.cmake
index d17e3e20e0..907881072f 100644
--- a/cmake/app/boilerplate.cmake
+++ b/cmake/app/boilerplate.cmake
@@ -86,10 +86,7 @@ set_property(DIRECTORY APPEND PROPERTY CMAKE_CONFIGURE_DEPENDS ${AUTOCONF_H})
 # that we need to work around until we upgrade to 3.13. Until then we
 # maintain a patched copy in our repo. Bug:
 # https://github.com/zephyrproject-rtos/zephyr/issues/11103
-set(PythonInterp_FIND_VERSION 3.4)
-set(PythonInterp_FIND_VERSION_COUNT 2)
 set(PythonInterp_FIND_VERSION_MAJOR 3)
-set(PythonInterp_FIND_VERSION_MINOR 4)
 set(PythonInterp_FIND_VERSION_EXACT 0)
 set(PythonInterp_FIND_REQUIRED 1)
 include(${ZEPHYR_BASE}/cmake/backports/FindPythonInterp.cmake)
```

it detect the version I want.

**To Reproduce**
Install a new version of python, for example in a virtualenv and make it default

**Expected behavior**
cmake should detect the enabled python configured, not a specific fixed version.

**Impact**
You might have installed all requirements using one version that cmake will not detect and build will fail.

**Additional context**
Some CI systems come with lots of python versions installed, this is confusing and doomed to fail if we pick a specific version.

ulfalizer, 2018-12-05:
I'm seeing this too, though it's kinda nice in that I get compatibility testing for free. ;)

SebastianBoe, 2018-12-05:
To save time, I'd prefer to await the CMake version bump. And then see if the problem persists ...

nashif, 2018-12-05:
> To save time, I'd prefer to await the CMake version bump. And then see if the problem persists ...

whose time? :-)
I spent like 2 hours trying to figure out why none of the installed python modules our scripts are using are not being found, I guess I am not the only one...

SebastianBoe, 2018-12-05:
Was able to reproduce, it seems that when python3.4 is on path it takes priority over any 'python' executable on path.

Will look into a fix.

PR #11907 (2018-12-07): cmake: FindPythonInterp: Prioritize 'python' from path the highest
Users expect that not just "some" compatible python is detected, but
also that it will be the 'python' executable from PATH, at least when
this executable is valid.

To this end rewrite FindPythonInterp to give this executable the
highest priority.

This fixes #11857
commit 9de5bc91 (2018-12-07): cmake: FindPythonInterp: Prioritize 'python' from path the highest

Users expect that not just "some" compatible python is detected, but
also that it will be the 'python' executable from PATH, at least when
this executable is valid.

To this end rewrite FindPythonInterp to give this executable the
highest priority.

This fixes #11857
closed 2018-12-07 by nashif