zephyrproject-rtos/zephyr #11879: samples/bluetooth/ipsp does not  reply when pings to local link nodes

nazeer22, 2018-12-05:
I am testing ipsp example (samples/bluetooth/ipsp) on nRF52840-dk. when i flashed i get the below 
log from the board console.

***** Booting Zephyr OS zephyr-v1.13.0-2407-g459ad673ba *****
ecc stack (real size 1100):     unused 216      usage 884 / 1100 (80 %)
[00:00:00.008,789] <inf> bt_hci_core.hci_vs_init: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.008,789] <inf> bt_hci_core.hci_vs_init: HW Variant: nRF52x (0x0002)
[00:00:00.008,789] <inf> bt_hci_core.hci_vs_init: Firmware: Standard Bluetooth controller (0x00) Version 1.13 Build 99
[00:00:00.009,185] <inf> bt_hci_core.bt_dev_show_info: Identity: cd:d1:26:ea:15:21 (random)
[00:00:00.009,185] <inf> bt_hci_core.bt_dev_show_info: HCI: version 5.0 (0x09) revision 0x0000, manufacturer 0x05f1
[00:00:00.009,216] <inf> bt_hci_core.bt_dev_show_info: LMP: version 5.0 (0x09) subver 0xffff
[00:00:00.249,847] <inf> net_config.net_config_init: Initializing network
**[00:00:00.250,061] <wrn> net_if.net_if_send_data: iface 0x2000aa80 is down
[00:00:00.250,091] <err> net_if.join_mcast_allnodes: Cannot join all nodes address ff02::1 (-5)**
[00:00:03.253,265] <inf> ipsp.init_app: Run IPSP sample
[00:00:03.254,852] <inf> ipsp.listen: Starting to wait

I followed the README.rst provided in the ipsp. 
I am able to scan the address using hcitool lescan and connecting with echo "connect " command and able to get the bt0 interface. but when i do a ping to the local link node usnig ping6 -I bt0 ff02::1, board should reply marked with (DUP!) which i am not getting.

So is there any way to know the ipv6 address of the board to that i could ping to my board directly.

Regards.

nazeer22, 2018-12-05:
> Hi,
> IPSP sample by default uses statically configured address: `2001:db8::1`, there's a description on how to use it further in the sample documentation: https://docs.zephyrproject.org/latest/samples/bluetooth/ipsp/README.html.
> Anyway, I've tested the sample with link-local multicast ping (`ff02::1`) and it worked for me as expected (tested on `nrf52840_pca10056`). This error from `join_mcast_allnodes` doesn't look pretty though.

Hi,
I followed the link  https://docs.zephyrproject.org/latest/samples/bluetooth/ipsp/README.html,
i tried to ping to the static address  2001:db8::1 also , but it also failed.

Below is the log from my observation.
1) # modprobe bluetooth_6lowpan
    # echo 1 > /sys/kernel/debug/bluetooth/6lowpan_enable
2) Connected the board to the uart console, below is the output from the console

***** Booting Zephyr OS zephyr-v1.13.0-2407-g459ad673ba *****
ecc stack (real size 1100):     unused 216      usage 884 / 1100 (80 %)
[00:00:00.008,819] <inf> bt_hci_core.hci_vs_init: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.008,819] <inf> bt_hci_core.hci_vs_init: HW Variant: nRF52x (0x0002)
[00:00:00.008,819] <inf> bt_hci_core.hci_vs_init: Firmware: Standard Bluetooth controller (0x00) Version 1.13 Build 99
[00:00:00.009,216] <inf> bt_hci_core.bt_dev_show_info: Identity: cd:d1:26:ea:15:21 (random)
[00:00:00.009,216] <inf> bt_hci_core.bt_dev_show_info: HCI: version 5.0 (0x09) revision 0x0000, manufacturer 0x05f1
[00:00:00.009,246] <inf> bt_hci_core.bt_dev_show_info: LMP: version 5.0 (0x09) subver 0xffff
[00:00:00.249,877] <inf> net_config.net_config_init: Initializing network
**[00:00:00.250,091] <wrn> net_if.net_if_send_data: iface 0x2000aa80 is down
[00:00:00.250,122] <err> net_if.join_mcast_allnodes: Cannot join all nodes address ff02::1 (-5)**
[00:00:03.253,265] <inf> ipsp.init_app: Run IPSP sample
[00:00:03.255,035] <inf> ipsp.listen: Starting to wait

3) Now i scanned for the BLE Address of board,using the bluetoothctl utility.

root@debian:~# bluetoothctl
[NEW] Controller 00:1B:DC:06:96:17 debian [default]
[NEW] Device 5A:6A:53:29:69:50 [LG] webOS TV UJ675V
[NEW] Device CD:D1:26:EA:15:21 Test IPSP node
[NEW] Device CE:96:AD:51:0B:0C Test IPSP node
[bluetooth]# scan on
Discovery started
[CHG] Controller 00:1B:DC:06:96:17 Discovering: yes
[CHG] Device CD:D1:26:EA:15:21 RSSI: -57
[CHG] Device 5A:6A:53:29:69:50 RSSI: -88
[CHG] Device B0:B4:48:CA:08:06 RSSI: -81
[CHG] Device B0:B4:48:CA:08:06 TxPower: 0
[CHG] Device CD:D1:26:EA:15:21 RSSI: -44
[CHG] Device CD:D1:26:EA:15:21 RSSI: -53
[CHG] Device CD:D1:26:EA:15:21 RSSI: -64
[CHG] Device CD:D1:26:EA:15:21 RSSI: -55

CD:D1:26:EA:15:21 is my board BLE Address 

4) echo "connect CD:D1:26:EA:15:21 2" > /sys/kernel/debug/bluetooth/6lowpan_control 
    and i was able to get the bt0 interface 
5) Then i added ip address add 2001:db8::2/64 dev bt0 to ping to the static address of the board.
    and when i do a ping6 2001:db8::2 , it failed . 

Is there any thing else i need to configure to ping to my board.

ntavish, 2018-12-05:
To test this, I was always running radvd, some documentation here: https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.iotsdk.v0.9.0/iot_sdk_user_guides_radvd.html

I'm not 100% sure if it's really needed though.

nazeer22, 2018-12-06:
Also i enabled the shell on nrf52840_pca10056 and checked the interface using net iface and it shows interface is down.

uart:~$ net iface

Interface 0x2000bdc0 (Bluetooth) [0]
====================================
Interface is down.

rlubos, 2018-12-07:
@nazeer22 I'm not an expert in terms of IPSP, but from experience I can tell that for BLE IPSP, network interface will be down until you have a valid BLE connection established. Once you connect to the board from your host, you should see something like this (it's what current master prints):
```
Interface 0x200107a0 (Bluetooth) [0]
====================================
Link addr : E3:88:DD:72:3A:F3
MTU       : 1280
IPv6 unicast addresses (max 3):
        2001:db8::1 manual preferred infinite
        fe80::e388:ddff:fe72:3af3 autoconf preferred infinite
IPv6 multicast addresses (max 4):
        ff02::1
IPv6 prefixes (max 2):
        <none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 35825
IPv6 retransmit timer    : 0
```
And typically when testing I follow the commands you posted.