zephyrproject-rtos/zephyr #11904: blinky example can't work on nucleo-f070rb and nucleo-f030r8 platform

qianfan-Zhao, 2018-12-06:
**Describe the bug**
samples/basic/blinky Example can't work on nucleo-f070rb and nucleo-f030r8 platform(the led can't blinking after download), but the hello_world example can work.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nucleo_f070rb ..
3. make
4. make flash

erwango, 2018-12-06:
Seems fixing BUS in stm32f0.dtsi fixes the issue:
```
 			gpioa: gpio@48000000 {
 				compatible = "st,stm32-gpio";
 				gpio-controller;
 				#gpio-cells = <2>;
 				reg = <0x48000000 0x400>;
-				clocks = <&rcc STM32_CLOCK_BUS_AHB2 0x00020000>;
+				clocks = <&rcc STM32_CLOCK_BUS_AHB1 0x00020000>;
```

Though, in that case console should not be working neither.
With the change console is still working...
There is definitly something strange here.

erwango, 2018-12-06:
AHB2 bus is actually not supported on F0 series.
Things (eg hello_world) should be working as AHB1 bus is activated by request from other subsystem.
Will submit a patch to fix this and add various check points to detect this kind of issues earlier in future

PR #11925 (2018-12-07): STM32: Fix clock bus for STM32F0 GPIO
Clock bus was not documented correctly in stm32f0.dtsi.
Failure was exposed since recent change in GPIO drivers using bus setting from device tree.

To avoid this kind of issue in future, add default case that return error when unsupported bus is requested from peripheral drivers. Add matching error checking in peripheral drivers.

Fixes #11904
commit f6b014e6 (2018-12-07): dts/arm: stm32f0: AHB2 but is not supported

AHB2 bus does not exist on STM32F0 series, replace with AHB1.

Fixes #11904
closed 2018-12-07 by nashif