zephyrproject-rtos/zephyr #11908: Power Manager does not handle K_FOREVER properly

baw-serafin, 2018-12-06:
**Describe the bug**
Power Manger will go into deep sleep if `sys_pm_policy_next_state` is called with K_FOREVER even if there is still work to be done.

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout v1.13
2. Build `/samples/subsys/power` with ` cmake -DBOARD=nrf52_pca20020 -GNinja ../..` (defaults)
3. run it on a thingy 52

**Expected behavior**
See App output on console:
```
***** Booting Zephyr OS zephyr-v1.13.0 ****

***OS Power Management Demo on arm****
Demo Description
Application creates Idleness, Due to which System Idle Thread is
scheduled and it enters into various Low Power States.
```

**Actual behavior**
Only `Entering Low Power state (3)` is printed. device is in deep sleep and no possibility to wake it up. `main` is never reached

**Impact**
Big we have to implement own Power Manager / Patch policy.c which is scary

**Screenshots or console output**
see above

**Environment (please complete the following information):**
 - OS: Debian Linux
 - Toolchain Zephyr SDK
 - `<v1.13.0>`

**Workaround**
The problem seems to be with the handling of `K_FOREVER` following diff will mitigate problem but probably has other issues:
```
diff --git a/subsys/power/policy.c b/subsys/power/policy.c
index ef204e1dde..5eb32965ec 100644
--- a/subsys/power/policy.c
+++ b/subsys/power/policy.c
@@ -72,7 +72,7 @@ int sys_pm_policy_next_state(s32_t ticks, enum power_states *pm_state)
 {
        int i;
 
-       if ((ticks != K_FOREVER) && (ticks < pm_policy[0].min_residency)) {
+       if ((ticks == K_FOREVER) || (ticks < pm_policy[0].min_residency)) {
                LOG_ERR("Not enough time for PM operations: %d\n", ticks);
                return SYS_PM_NOT_HANDLED;
        }

```

pizi-nordic, 2018-12-06:
IMHO the problems is in the:
```
/* This Function decides which Low Power Mode to Enter based on Idleness.
 * In actual Application this decision can be made using time (ticks)    
 * And RTC timer can be programmed to wake the device after time elapsed.
 */                                                                      
static int pm_policy_get_state(s32_t ticks)
{
        /*
         * Power Management Policy:
         * 0 - 60 Sec:  Enter Constant Latency Mode
         * 61 - 120 Sec: Enter Low Power state
         * 121 and above : Enter Deep Sleep state
         */
        if ((ticks != K_FOREVER) && ticks <= LPS_MAX_THR) {
                return SYS_POWER_STATE_CPU_LPS;
        } else if ((ticks > LPS_MAX_THR) &&
                        (ticks <= LPS_1_MAX_THR)) {
                return SYS_POWER_STATE_CPU_LPS_1;
        } else {
                return SYS_POWER_STATE_DEEP_SLEEP;
        }
}
```
We never should select SYS_POWER_STATE_DEEP_SLEEP basing on the expected sleep time as this maps to "System OFF" in our architecture. Since we enter the mode, we cannot be woken-up in the way Zephyr expects (exit from System OFF looks like reset).

Also the mapping of the Zephyr states to the SoC settings in not fortunate:
```
enum power_states {                                                                                                                                                                           
        SYS_POWER_STATE_CPU_LPS,       /* CONSTLAT Task Activited */                                                                                                                          
        SYS_POWER_STATE_CPU_LPS_1,     /* LOWPWR Task Activated */                                                                                                                            
        SYS_POWER_STATE_CPU_LPS_2,     /* Not supported*/                                                                                                                                     
        SYS_POWER_STATE_DEEP_SLEEP,    /* System Off */                                                                                                                                       
        SYS_POWER_STATE_DEEP_SLEEP_1,  /* Not Supported */                                                                                                                                    
        SYS_POWER_STATE_DEEP_SLEEP_2,  /* Not Supported */                                                                                                                                    
                                                                                                                                                                                              
        SYS_POWER_STATE_MAX             /* Do nothing */                                                                                                                                      
};                                                                                                                                                                                                                                                                                                                                                                                     ```

The CONSTLAT and LOWPWR settings are affecting only the PPI interconnect. In general you should not use the CONSTLAT if the "hardware" you crafted in the PPI does not need that. By setting PPI into CONSTLAT mode when it is not needed, you are increasing power usage by 500 uA.

pizi-nordic, 2018-12-06:
@carlescufi: FYI

baw-serafin, 2018-12-06:
@pizi-nordic 
> IMHO the problem is in the:
> 
> ```
> /* This Function decides which Low Power Mode to Enter based on Idleness.
>  * In actual Application this decision can be made using time (ticks)    
>  * And RTC timer can be programmed to wake the device after time elapsed.
>  */                                                                      
> static int pm_policy_get_state(s32_t ticks)
> ```

I thought so too, but I think that code is not reached with `samples/subsys/power`. The power.c in which this code lies is only in `sample/boards/nrf52/power_mgr` is it not? Or do I miss something?

pizi-nordic, 2018-12-06:
@baw-serafin: You are right, the problem you observe is different, however the `pm_policy_get_state()` have to be fixed too. The code selecting power state is in the policy (`sys_pm_policy_next_state()`).

Calling the `sys_pm_ctrl_disable_state(SYS_PM_DEEP_SLEEP)` as early as possible should workaround the problem.

ramakrishnapallala, 2019-02-18:
@pizi-nordic is this taken care in your reworked commits?

pizi-nordic, 2019-02-19:
There were some fixes in the polices, however I do not know if this particular problem is solved. I was not able to reproduce this issue on my setup.

pizi-nordic, 2019-02-19:
This is loosely related to https://github.com/zephyrproject-rtos/zephyr/issues/10524.
IMHO the deep sleep states should not be selected by OS policies unless requested.

ramakrishnapallala, 2019-02-25:
@pizi-nordic since you are reworking the sample and PM api's i am adding you the bug.

galak, 2019-03-26:
@pizi-nordic can you look at this again, is this an issue for 1.14?

pizi-nordic, 2019-03-27:
The /samples/subsys/power sample was removed. The samples/boards/nrf52/power_mgr, which was updated to match recent development in power management area does not support the nrf52_pca20020 board as it lacks buttons and LEDs required by the example. So this bug is no longer valid.

Unfortunately root cause of the behaviour described in this bug is lack of some PM features. But these are covered in https://github.com/zephyrproject-rtos/zephyr/issues/14308 and https://github.com/zephyrproject-rtos/zephyr/issues/14306

IMHO we should close this at the moment.

baw-serafin, 2019-03-27:
I think the problem is not pc20020 specific test should be fairly easy. Use a device which loads a driver that waits an a lock (or similar) with K_FOREVER while having power management enabled. Device should not go into deepsleep. If it does it is very broken. As my code base is still 1.13 I can't test myself.

baw-serafin, 2019-03-27:
simething like:
```
static int test_device_init(struct device *dev)
{
     //initalize mutex
      //start delay work -> unlock mutex in 1s
      //lock mutex
      //wait mutex K_FOREVER
      printk("SUCCESS");
}

should do the trick (sorry for pseudo code)

pizi-nordic, 2019-03-27:
@baw-serafin: Yes, the problem is a generic one. However it is not so easy to fix. At the moment system does not have any information about state of the devices. So there is no simple way to prevent the shutdown. We can of course scan all devices and check if they are busy, but this will be an overkill (note that such scan would be needed on every idleness). These issues was already identified:

> Unfortunately root cause of the behaviour described in this bug is lack of some PM features. But this is covered in #14308 and #14306

However I do not see any possibility to have them properly fixed in short term due to amount of work needed in order to get something that will be easy to use.

baw-serafin, 2019-03-27:
@pizi-nordic  I agree it is a hard problem. I can only speak for the nrf52 platform as I don't know the rest.

I think it is a BUG that power_manager can go into deep-sleep as long as there is no automatic wakeup and ram retention this is unsuspected and will to lead to random "crashes" . But If you think the other bugs are explicit enough. You can close this one. But I think only  #14306 applies and it should be more clear that this will lead to dangerous behaviour. And i think it should be a bug not an enhancement ;)

pizi-nordic, 2019-03-27:
@baw-serafin: It think that #14306 will be still enhancement as it is hard to find bug in non-existing feature ;-). [The sad fact is that power management is mostly missing in Zephyr. Still the application have to handle most of the stuff].

Regarding this issue I could only propose a workaround: We can disable PM until application does not notify that it is ready to handle all events. What do you think about such approach?

baw-serafin, 2019-03-27:
@pizi-nordic that would be better. Going into a sleep state while bootup makes no sense anyway.

I still see a problem that the application also will not be able to handle a K_FOREVER signal. What should that even mean? I think there should be some kind of logic in the way of:

if (ticks == K_FOREVER) {
      ticks = CONFIG_IDLE_POLL_RATE;
}

Because nobody want's to sleep autonomic for FOREVER. If one has a device which is only triggered by a button or similar the application can go to deepsleep actively in all other cases it would be a very bad bug

pizi-nordic, 2019-03-27:
The problem is that drivers are using K_FOREVER to wait for interrupt (which is a bit silly - we should have timeout there) and PM assumes that any interrupt is able to wake-up the chip form any sleep mode which is simply not true. As result the application have to do a kind of micromanagement using sys_pm_force_power_state() and sys_pm_ctrl_enable/disable_state().

> Going into a sleep state while bootup makes no sense anyway.

It has a lot of sense if you have external sensor which needs plenty of time for initialization.

pabigot, 2020-05-26:
While the current implementation does have this problem, it can be avoid by adding code like this to the application:
```
sys_pm_ctrl_disable_state(SYS_POWER_STATE_DEEP_SLEEP_1);
```

Based on the presence of a simple workaround does this still qualify as a medium bug?  @carlescufi

baw-serafin, 2020-05-26:
Is it possible to make this default on nrf hardware? A new user will never figure this out

mnkp, 2020-05-27:
> While the current implementation does have this problem, it can be avoid by adding code like this to the application:
> 
> ```
> sys_pm_ctrl_disable_state(SYS_POWER_STATE_DEEP_SLEEP_1);
> ```

System power manager can schedule entry into one of the sleep states even before we reach main() and user has a chance to call `sys_pm_ctrl_disable_state()` function. While calling the function helps it is not really a foolproof workaround.

pabigot, 2020-05-27:
A pre-kernel SYS_INIT function could safely disable the feature per-application.

I wouldn't support having the default behavior of a system API like this differ between platforms.

Hopefully we can get the folks who depend on this behavior to review #21118 and get the defaults changed.

pabigot, 2020-09-16:
Since this refers to a sample that is no longer in the tree, and the replacement sample has instructions for how to work around this problem, I'm going to close this.  The new power management solution will not have this problem.

closed 2020-09-16 by pabigot