zephyrproject-rtos/zephyr #11912: net: SimpleLink: Create real cross-platform port

derchrismakaio, 2018-12-06:
Dear All,

I'm trying to implement the CC3220 WiFi driver and noticed that it is not really made for Zephyr. I mean, the port is totally dependent on running at TI's LaunchPad.
I'll create my own port now, somebody interested and willing to help?

GAnthony, 2018-12-06:
> I mean, the port is totally dependent on running at TI's LaunchPad.

@derchrismakaio, can you please clarify which elements of the WiFi driver are launchpad dependent?

I note that the Zephyr SimpleLink WiFi driver was also running on the 96boards-based HopeRun board.
See: https://www.youtube.com/watch?v=84AarmVjeUI
and http://hihope.org/product/Uranus

derchrismakaio, 2018-12-06:
@GAnthony Maybe I'm completely on the wrong track, but for example the
`CC3220SF_LAUNCHXL_init();` in driver's simplelink_support.c

cc_pal.c seems to "hardcode" SPI settings

Don't get me wrong, most of the work has been done by you. It's just the SPI HAL which seems to be sort of "hard"

GAnthony, 2019-02-13:
> @GAnthony Maybe I'm completely on the wrong track, but for example the
> `CC3220SF_LAUNCHXL_init();` in driver's simplelink_support.c
> 
> cc_pal.c seems to "hardcode" SPI settings
> 
> Don't get me wrong, most of the work has been done by you. It's just the SPI HAL which seems to be sort of "hard"
> 

   The SimpleLink host driver uses a dedicated SPI line to the NWP, and so writing a Zephyr SPI driver for the host driver wasn't a requirement (or desirable).  

> Also, I don't really get the relation between CONFIG_HAS_CC3220SDK and CONFIG_SIMPLELINK_HOST_DRIVER. Maybe it's mean to differ between the basic host driver and the launchxl-specific implementation? But looking at ext/hal/ti/simplelink/CMakeLists.txt shows that only if both are Y, the important files (like SPI.c) are included

It's historical:  CONFIG_HAS_CC3220SDK was originally created to enable Zephyr applications to use the SDK driverlib functions (and this was done well before the host driver was integrated).    Then, WiFi support was added later, and the CONFIG_SIMPLELINK_HOST_DRIVER variable was created to enable pulling in the host driver.

closed 2020-10-04 by github-actions[bot]