zephyrproject-rtos/zephyr #11916: ISR table (_sw_isr_table) generation is fragile and can result in corrupted binaries

mbolivar, 2018-12-06:
**Describe the bug**

If `CONFIG_GEN_ISR_TABLES=y`, drivers can use `IRQ_CONNECT()` to register an ISR with the arch IRQ handling core. `IRQ_CONNECT()` parameters are used to initialize a static variable stored in a special section named `.intList`. This data includes the address of the ISR function.

There is a global `_sw_isr_table` variable. The goal is to fill it with the `IRQ_CONNECT()` data.
During the build, Zephyr is linked for the first time, producing an ELF binary with a weak value for `_sw_isr_table` that has the correct size (`CONFIG_NUM_IRQS`) but garbage data, along with a final `.intList` section that has the data we want.

The `.intList` data are extracted into a blob, which `gen_isr_tables.py` uses to create the real `_sw_isr_table` in a generated C file. This file and the other Zephyr objects are linked again. The real `_sw_isr_table` overrides the weak symbol. This is the final ELF file. As long as the first and second link result in ISRs at the same addresses, everything works fine.

The problem is that this assumption is not in general valid. In particular, on RISC-V, unusually aggressive linker relaxation (https://www.sifive.com/blog/all-aboard-part-3-linker-relaxation-in-riscv-toolchain) -- whose implementation details vary by toolchain version -- can and do produce different binaries depending on random things.

The one that bit me in particular was that the RV32M1 toolchain (for a RISC-V SoC being added in https://github.com/zephyrproject-rtos/zephyr/pull/11853) chooses to emit a compressed instruction when loading the address of a strong symbol, but not a weak one.

Now, `_sw_isr_table` *itself* is a weak symbol in the first link and a strong one in the second. This SoC port involved a multi-level interrupt driver, which needs to get the addresses of level 2 interrupts from `_sw_isr_table`, and thus needs to load the `_sw_isr_table` address. Due to the linker relaxation details on my toolchain, the second binary had different-sized code for getting this address, bumping the addresses of the ISRs themselves backwards by a couple of bytes, resulting in a corrupted binary.

I was able to work around this for my toolchain, but issues like this are likely to happen again -- potentially  as we add non-GCC toolchains as well. This issue asks that the `IRQ_CONNECT()` and `gen_isr_tables.py` implementation be re-worked to be less fragile in the face of different emitted code in the first and second link.

A RISC-V expert I spoke to recommended putting the symbols of the ISRs in `_sw_isr_table` instead of their addresses (presumably with declarations before the table generated by `gen_isr_tables.py`). This would seem to require reworking `IRQ_CONNECT` to save the names of the ISRs rather than just their addresses as well.

**To Reproduce**

See this gist for an illustration of the behavior related to weak and strong symbols:

https://gist.github.com/mbolivar/e2704c48e92f13a0a731198603093dbc

In particular, loading the upper bits of the `global_symbol` address uses a 2 byte compressed `lui` instruction for a strong symbol:

https://gist.github.com/mbolivar/e2704c48e92f13a0a731198603093dbc#file-linker-relaxation-on-rv32m1-toolchain-L63

And a 4-byte non-compressed instruction for a weak symbol:

https://gist.github.com/mbolivar/e2704c48e92f13a0a731198603093dbc#file-linker-relaxation-on-rv32m1-toolchain-L74

Both binaries are correct and semantically equivalent, but they differ in size. Instruction addresses for any code after this function would be offset by 2B between the two, even if nothing else changed.

**Expected behavior**

Generated IRQ handling code should correctly access driver-specified ISRs.

**Impact**

Zephyr binaries on RISC-V can be corrupted due to "phase of the moon" implementation details in the linker.

**Screenshots or console output**

See above gist.

**Environment (please complete the following information):**
 - OS: any
 - Toolchain (e.g Zephyr SDK, ...): RISC-V GCC toolchain for RV32M1 SoC. Note Zephyr SDK RISC-V toolchain behaves differently! (It emits uncompressed instructions for both links)
 - Commit SHA or Version used: any since gen_isr_tables.py was added

**Additional context**

N/A

jim-wilson, 2018-12-06:
I'm not able to reproduce with any toolchain I have access to.  Checking your toolchain, I see that you are using ancient versions of the tools from mid 2017.  The tools weren't stable until about January 2018.

```
rohan:2037$ riscv32-unknown-elf-gcc --version
riscv32-unknown-elf-gcc (GCC) 7.1.1 20170509
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

rohan:2038$ riscv32-unknown-elf-ld --version
GNU ld (GNU Binutils) 2.28.0.20170505
Copyright (C) 2017 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
rohan:2039$ 
```

I wouldn't recommend using anything before FSF gcc-7.3 and FSF binutils-2.30.  Those are the earliest releases that are stable enough to build an entire linux distro.

I suspect that the linker problem you are seeing was fixed sometime last year, possibly before the port was contributed to the FSF, which makes it difficult to figure out what exactly changed that fixed this.

By the way, your "source file" releases contain pdfs, and appear to be identical to your documentation release.  That is probably a bug on your web site.

andrewboie, 2018-12-06:
It's not quite clear to me from reading this whether this is a toolchain or a zephyr problem. Can we establish whether generating different code depending on whether weak symbols are used or not is permissible behavior, before we rework the interrupt code?

jim-wilson, 2018-12-06:
The compiler doesn't know if a weak has zero or non-zero address, so must emit a generic sequence.  If a non-weak is not defined in the module where it is used, then this must also be a generic sequence.  I'd expect these to be the same.

A non-weak defined in the same module where it is used might be optimized by the compiler.  But whether this affects instruction length depends on the hardware addressing modes, and the code models  and ABIs in use by the compiler.  It probably ends up the same length anyways.

If you are doing link-time-optimization, then a reference to a weak may be optimized away, or maybe optimized to a trap instruction if your target doesn't support reads from a 0 address.  This is very likely to change code size.

The RISC-V port does aggressive linker relaxation by default, which like link-time-optimization, can change code size.  The specific example shown in this bug report I'd call a toolchain bug though, as you should have gotten the same code at the end, and I do get the same code size with all supported toolchains versions I tried.

The testcase given here defines the weak variable in the same module where it is referenced though.  If you change the example to make it an extern, then it gets more interesting.  I get
```
lui	a5,0x0
lw	a0,0(a5)
```
which could be optimized at link time to just "lw a0,0(x0)".  We may add a linker relaxation for this some day, and then you could have a problem again, as the code size would be different for the weak and non-weak references.

Incidentally, the RISC-V linker relaxation can be turned off temporarily.  You can do
```
.option push
.option norelax
...
.option pop
```
and the section in the middle won't be subject to linker relaxation.  This could be used to work around this problem.  This is a lot easier to use in an assembler file than a C file, but you try using volatile asms to put this in C file.  You can also compile the C file with -mno-relax if you have an up-to-date compiler, or you can just use -Wa,-mno-relax with an older toolchain.

mbolivar, 2018-12-06:
> I wouldn't recommend using anything before FSF gcc-7.3 and FSF binutils-2.30. Those are the earliest releases that are stable enough to build an entire linux distro.

Hi there @jim-wilson! I unfortunately don't control the toolchain and cannot implement this recommendation, even though I wish I could.

> It's not quite clear to me from reading this whether this is a toolchain or a zephyr problem. Can we establish whether generating different code depending on whether weak symbols are used or not is permissible behavior, before we rework the interrupt code?

I think this is a case of Zephyr's two-stage link being a bit too chummy with the implementation.

Generating different code depending on whether weak symbols are used is perfectly permissible as far as I can tell. The RISCV binutils maintainer (Palmer Dabbelt) I spoke with said that in general it's not safe to rely on two links even with slightly different code producing the same results. He told me Linux was also doing two-pass linking and also had to make similar changes.

mbolivar, 2018-12-06:
> and the section in the middle won't be subject to linker relaxation. This could be used to work around this problem. This is a lot easier to use in an assembler file than a C file, but you try using [...]

I am aware of and did consider that. As I said, I was able to work around this for this toolchain.

My concern is to avoid future driver authors running into issues like this based on what to me seems like invalid assumptions in the Zephyr build system.

andrewboie, 2018-12-07:
> I spoke with said that in general it's not safe to rely on two links even with slightly different code producing the same results. He told me Linux was also doing two-pass linking and also had to make similar changes.

Hmm, fair enough.
For the ISR tables, it won't be too bad to change it to save symbols instead of addresses, I think that will fix this.

We have some other stuff on x86 where this becomes more of a problem. Building the GDT and IDT at build time can't be done with symbols since its a binary blob that gets built, not C code, and due to way the addresses are split around in various ways in these descriptor data structures, you can't statically define them in C. But perhaps since this is just x86 we can leave it alone.

I think the privileged stack generation mechanism used on ARM/ARC might also rely on memory addresses staying the same, but I need to double-check.

SebastianBoe, 2018-12-07:
Perhaps instead of using weak symbols we should use different object files in different linker passes.

palmer-dabbelt, 2018-12-07:
I might have said this wrong during the summit, but the two-pass link in Linux was doing something different: it didn't expect stable symbol addresses, but instead used weak undefined symbols that later become defined.  The Linux code was legal, it was just triggering a linker bug.

I can't figure out a better way to deal with the harder problems :)

andrewboie, 2018-12-07:
> Perhaps instead of using weak symbols we should use different object files in different linker passes.

I tried this at one point. Unfortunately, the objects would appear in different places in the binary, which broke stuff. They have to be guaranteed to be in the same place or it falls over. Might still be possible but everything would have to be linked in the same order. This might be the best route actually, see below:

> For the ISR tables, it won't be too bad to change it to save symbols instead of addresses, I think that will fix this, since the output of the scripts is just more C code which gets built.

Hmm. I may have spoke to soon. We can use symbols for the ISR addresses, but it gets more complicated when it comes to the argument passed to the ISR, this gets stored too and can be anything, a constant, a symbol, NULL, etc.

andrewboie, 2018-12-07:
I think it's probably just simplest to revert 140863f6a753f3ab169e0faee79463e977c1364f
and go back to using .gnu.linkonce.

mbolivar, 2018-12-11:
> I think it's probably just simplest to revert [140863f](https://github.com/zephyrproject-rtos/zephyr/commit/140863f6a753f3ab169e0faee79463e977c1364f)
> and go back to using .gnu.linkonce.

@andrewboie:

I saw this and appreciate the fix for the issue with my (ancient) toolchain.

More generally, would it be worth it to do a post-processing step on the final zephyr binary to make sure that ISR addresses didn't change, and at least fail fast if they did?

jasonedn, 2020-04-11:
I have similar issue (#24291) on cortex. The generated source code isr_tables.c is correct, but _sw_isr_tables in the final linked image is not correct. When button gpio interrupt occurred, it entered the default spurious handler instead of the button handler.