zephyrproject-rtos/zephyr #11922: ADC generic driver missing features: calibration, reference voltage value

pabigot, 2018-12-07:
The Nordic SAADC peripheral has a calibration feature that's supposed to be initiated before the first use.  The existing generic ADC driver does not have a way to initiate this.  Since configuring ADC through the "generic" API requires including the SOC headers anyway, I'm just calibrating manually.

Also there's an option to use the internal reference voltage, but no way to fetch the value of that voltage for whatever board you're running on so the measurement in ADC counts can be converted to a voltage.   The value isn't even in the nrfx HAL as far as I can tell.

carlescufi, 2018-12-07:
@anangl was the person leading the ADC API rework that took place recently.

Qbicz, 2018-12-10:
Thumbs up, I would be happy to see SAADC calibration feature added.

pabigot, 2018-12-26:
When somebody comes to support retrieving the internal reference voltage here's what I've found:

* For `CONFIG_ADC_NRFX_ADC` it's 1200 mV
* For `CONFIG_ADC_NRFX_SAADC` it's 600 mV
* For ARC Designware ADC the information is not publicly available
* For Intel Quark D2000 it's either 3.3V or the battery voltage
* For Kinetis (MCUX_ADC16) there is no internal reference voltage, although ADC_REF_INTERNAL is the only permitted configuration.  The actual reference voltage is selected from one of two external choices, possibly constrained by the PCB, and fixed via Kconfig.
* For Atmel SAM AFEC it's either 3.3V or 5.0V.  For `sam_e70_xplained` it appears to be 3300 mV.
* No other platforms implement this API.

Looking at the huge variation in ADC support across just these few platforms I suspect it would be wiser to reduce the complexity of the ADC API to a bare-bones set of operations that do not attempt to provide a generic API for things like channel configuration, acquisition time, resolution, or sequencing.

Or just remove it and accept that doing ADC requires a platform-specific driver with no generic abstraction, which would probably be a more maintainable and simpler approach.

pabigot, 2019-08-01:
Looks like calibration and reference voltage APIs are now both present, so if there's a problem with them that can be a new issue.

closed 2019-08-01 by pabigot