zephyrproject-rtos/zephyr #11928: nRF UART nrfx drivers (nRF UARTE 0) won't build

oxymoron87, 2018-12-07:
**Describe the bug**
Can't build project for `set(BOARD nrf52840_pca10056)` with `CONFIG_UART_0_NRF_UARTE=y`. Build process fails on undeclared symbol.
`[build] ../zephyr/drivers/serial/uart_nrfx_uarte.c:581:5: error: 
 DT_NORDIC_NRF_UARTE_UART_0_CURRENT_SPEED' undeclared here (not in a function); did you mean 'DT_NORDIC_NRF_UART_UART_0_CURRENT_SPEED'?`
`[build]      DT_NORDIC_NRF_UARTE_UART_##idx##_CURRENT_SPEED,\`
`[build]      ^`
`[build] ../zephyr/drivers/serial/uart_nrfx_uarte.c:689:2: note: in expansion of macro` `'UART_NRF_UARTE_DEVICE'`
`[build]   UART_NRF_UARTE_DEVICE(0);`
`[build]   ^~~~~~~~~~~~~~~~~~~~~`
`[build] ../zephyr/drivers/serial/uart_nrfx_uarte.c:597:4: error:` `'DT_NORDIC_NRF_UARTE_UART_0_BASE_ADDRESS' undeclared here (not in a function); did you` `mean 'DT_NORDIC_NRF_UART_UART_0_BASE_ADDRESS'?`
`[build]     DT_NORDIC_NRF_UARTE_UART_##idx##_BASE_ADDRESS,        \`
`[build]     ^`
`[build] ../zephyr/drivers/serial/uart_nrfx_uarte.c:689:2: note: in expansion of macro` `'UART_NRF_UARTE_DEVICE'`
`[build]   UART_NRF_UARTE_DEVICE(0);`
`[build]   ^~~~~~~~~~~~~~~~~~~~~`
etc.

**To Reproduce**
Steps to reproduce the behavior:
1. set `CONFIG_UART_0_NRF_UARTE=y` in prj.conf
2. mkdir build; cd build
3. cmake -DBOARD=nrf52840_pca10056
3. make
4. See error

**Additional context**
Symbols are generated to
`./build/zephyr/include/generated/generated_dts_board.conf` (DT_NORDIC_NRF_UART_**UART**_0_CURRENT_SPEED=115200)
 but for UART and not for UARTE even if UARTE is selected.

**Impact**
Can't use nRF UARTE instead of nRF UART

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: GNU ARM GCC
 - ZephyrOS commit: 48f493208baf4d7e6d8f72b1473889b2cc9fcbcb

jarz-nordic, 2018-12-07:
This is philosophical problem, what shall be done in DT and what in Kconfig. 

We agreed that Zephyr will be Linux-like so in Kconfig you set a driver to be compiled and in DT peripheral to be used.

Please refer to issue: #11792 

To fix it simply edit file: 
`zephyr/boards/arm/nrf52840_pca10056/nrf52840_pca10056.dts`
you need to change line 92:
`compatible = "nordic,nrf-uart";` to `compatible = "nordic,nrf-uarte";`

closed 2018-12-07 by jarz-nordic

derchrismakaio, 2018-12-13:
@jarz-nordic could #11932, #11792 or this one please stay open? I think this is a real problem, especially keeping newbies in mind. I always wonder why there aren't more hints in code. In this case, for example, it could be something like

```
#if !defined(DT_NORDIC_NRF_UARTE_UART_0_CURRENT_SPEED) && defined(DT_NORDIC_NRF_UART_UART_0_CURRENT_SPEED)
#error Please set UART0 compatible = "nordic,nrf-uarte" in DTS
#endif
```
in uart_nrfx_uarte.c

Again, this is just an example and I think that it's not good to place it there because it statically checks for UARTE0 (instead of being generic like UART_NRF_UARTE_DEVICE(idx))

Should I create an Improvement issue for "Help newbies with compiler macros"? Eg, what about adding an extra build step which calls a "config_check.h" afterwards and everyone who wants to add checks could just include for example "config_check_nrfx_uarte.h" there and define checks in that header file?

jarz-nordic, 2018-12-13:
@derchrismakaio : I am not an expert to say if this is good idea or not.
I would ask: @anangl , @galak  or @Mierunski , they were working a lot on Device Tree.

CodingGhost, 2020-04-09:
I am a newbie here and have to agree @chris-makaio.
I had no Idea what to do.