zephyrproject-rtos/zephyr #11932: CMake error when deselecting "UART Port 0 Driver type"

oxymoron87, 2018-12-07:
**Describe the bug**
CMake error when deselecting "UART Port 0 Driver type" in Device Drivers -> Serial Drivers -> nRF UART nrfx drivers (through menuconfig).

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf52840_pca10056 ..
3. ninja menuconfig
4. deselect Device Drivers -> Serial Drivers -> nRF UART nrfx drivers -> UART Port 0 Driver type
3. ninja
4. See error

**Screenshots or console output**
`CMake Error at zephyr/CMakeLists.txt:674 (message):`
`  The Zephyr library 'drivers__serial' was created without source files.`
`  Empty (non-imported) libraries are not supported.  Either make sure that`
`  the library has the sources it should have, or make sure it is not created`
`  when it has no source files.`

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: GNU ARM GCC
 - Commit 48f493208baf4d7e6d8f72b1473889b2cc9fcbcb

jarz-nordic, 2018-12-07:
@oxymoron87 :In my opinion it is expected behavior. You've selected that you want to use serial but you did not choose a driver. It seems to be related to your other issue: #11928. Do you still consider this PR as an issue?

jarz-nordic, 2018-12-12:
@oxymoron87 : Please let me know if this is still an issue for you or it can be closed?

oxymoron87, 2018-12-12:
I just thought that UI (menuconfig) should not let me to do so, but that's my opinion.  Not an issue anymore.

carlescufi, 2018-12-12:
@oxymoron87 unfortunately there's a bit more coupling between Device Tree and Kconfig than it might be apparent. We are working towards improving the user experience, and we are aware of this issue, thanks for letting us know!

closed 2018-12-12 by carlescufi