zephyrproject-rtos/zephyr #11967: nrfx SPI driver blocks indefinitly when transfering

NZSmartie, 2018-12-08:
The SPI driver for Nordic's nRF5x chips blocks indefinitely when performing a transfer.

**To Reproduce**
Steps to reproduce the behaviour:
1. Create a project targeting any nRF5x SOCs, (i.e. `nrf51_ble400`, ..., `nrf52_pca10040`)
1. Enable a SPI driver. 
   For example, here is a snippit from defconfig from enabling `SPI_0`

   ```Makefile
   # Enable Nordic SPI on bus 0
   CONFIG_SPI=y
   CONFIG_SPI_0_IRQ_PRI=10
   CONFIG_SPI_0=y
   CONFIG_SPI_NRFX=y
   CONFIG_SPI_0_NRF_SCK_PIN=0
   CONFIG_SPI_0_NRF_MOSI_PIN=1
   CONFIG_SPI_0_NRF_MISO_PIN=2
   ```
1. In the application, Get the SPI device and initiate a `spi_write(...)`
   ```c
   #include <spi.h>

   // ...

   // Setup the SPI
   struct device *spi = device_get_binding(CONFIG_SPI_0_NAME);
   struct spi_config cfg = {
       .frequency = 2000000,
       .operation = SPI_OP_MODE_MASTER | SPI_TRANSFER_MSB | SPI_WORD_SET(8),
   };

   // Write some data
   u8_t data[] = {1,2,3,4};
   struct spi_buf tx_buf = {.buf = data, .len = 4};
   struct spi_buf_set tx_bufs = {.buffers = &tx_buf, .count = 1};
   spi_write(spi, cfg, &tx_bufs);
   ```

**Expected behavior**
The SPI driver does not block and execution flow of the program at `spi_write(...)` and continues as expected.

**Impact**
The SPI and SPIM driver for Nordic's nRF5x series SOCs will block indefinitely the moment a transfer is invoked. Typically through `spi_write`

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04 (Windows Subsystem for Linux)
 - Toolchain 
   - Zephyr v1.13.0, 
   - GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release)

**Additional context**
Add any other context about the problem here.

NZSmartie, 2018-12-08:
I have identified what may be the cause of the block on line 153 of `spi_nrfx_spi.c`

https://github.com/zephyrproject-rtos/zephyr/blob/62c8fee2ecadb48b84b6115567db3422ab5b94c3/drivers/spi/spi_nrfx_spi.c#L134-L166

If the SPI xfer is a success, the function immediately returns without clearing the semaphore associated with the driver's context.

I've noticed that the same pattern is used with [`spi_nrfx_spim.c`](https://github.com/zephyrproject-rtos/zephyr/blob/62c8fee2ecadb48b84b6115567db3422ab5b94c3/drivers/spi/spi_nrfx_spi.c#L153-L157) as well

NZSmartie, 2018-12-08:
Would like to CC: @anangl on this as well

anangl, 2018-12-10:
> I have identified what may be the cause of the block on line 153 of `spi_nrfx_spi.c`
>
> [..]
>
> If the SPI xfer is a success, the function immediately returns without clearing the semaphore associated with the driver's context.

It is intentional. `transfer_next_chunk` is expected to return immediately when `nrfx_spi_xfer` signals success because this success means only that the transfer of a chunk has been started. After the transfer is completed, `event_handler` will be called, and it will in turn call `transfer_next_chunk` again. Then if there is nothing more to transfer (`chunk_len` is `0`), `nrfx_spi_xfer` will not be called and the function will return with a prior call to `spi_context_complete` (as it does in case of an error).

What you observe is most likely not caused by a bug in the driver. It's rather a wrong configuration of the hardware. I suppose that the SCK line is driven by something else and this results in an undefined behavior of the SPI peripheral. I can replicate the problem on the pca10056 board using the configuration you provided. But on this board pin 0 (which you configure to be the SCK line) is connected to the external LFCLK oscillator (same story with pin 1). After I configure this line to pin 3 for instance, the transfer is performed as expected.

galak, 2018-12-13:
Re-open if you feel this isn't related to HW.

closed 2018-12-13 by galak