zephyrproject-rtos/zephyr #11976: APIs that support a callback should provide both the device pointer and a generic pointer

pabigot, 2018-12-09:
I propose both a pointer to the associated device and a generic pointer should be provided consistently for all device API callbacks.

I'm running into an issue right now with `gpio_callback` where what comes in is the GPIO device `port` which I need to change the interrupt level sense, a pointer `cb` to a callback structure that lives within a sensor's driver data structure, and a bit-mask of pins associated with the callback.

From that I can get the sensor's `dev->driver_data` pointer by:
```
	struct sht3xd_data *drv_data =
		CONTAINER_OF(cb, struct sht3xd_data, gpio_cb);
```

But I can't get the sensors `dev->config->config_info` pointer which has the data that tells me which pin is which, because the base device pointer to the sensor device has been erased.  (Yes, I may share a callback with multiple pins: the API permits/encourages this by providing a mask rather than a pin.)

As a workaround I'm going to have to store a copy of the `dev->config->config_info` pointer in the `sht3xd_data` structure.  Had the `gpio_callback_handler_t` taken a generic pointer instead of a pointer to the callback, I could have passed it the sensor device pointer from which I could get both the callback structure and the configuration pointer.  People without this need could have passed the pointer to the callback structure, and have what they have today.

A callback for a device should always be given include the device for which it's a callback for consistency across APIs where this usually necessary, and for future-proofing for API enhancements where it may become necessary.  The API design should not assume that the device pointer can be fetched by doing offset manipulations based on a user-provided pointer.

Further, callbacks should always be passed a user-provided generic pointer rather than a pointer to the callback object, for the same reason of allowing flexibility and not making developers try to reconstruct it from a pointer to an embedded member.

pabigot, 2018-12-09:
@nordic-krch @carlescufi @MaureenHelm @mnkp @anangl This issue is intended as the basis for a more visible discussion following up on [this comment](https://github.com/zephyrproject-rtos/zephyr/pull/8340#discussion_r236914015) discussed in the API call last week.

mnkp, 2018-12-10:
> This issue is intended as the basis for a more visible discussion following up on [this comment](https://github.com/zephyrproject-rtos/zephyr/pull/8340#discussion_r236914015) discussed in the API call last week.

The GPIO callback is defined as
```
typedef void (*gpio_callback_handler_t)(struct device *port,
					struct gpio_callback *cb,
					u32_t pins);
```
and the limitations of this design were described in the the [answer](https://github.com/zephyrproject-rtos/zephyr/pull/8340#discussion_r237095779) to the above comment.

Had GPIO callback passed a generic user_data pointer as a parameter it would be sufficient to get access to all the required data. Solving the problem described in this issue doesn't require passing both the device pointer and a generic pointer in the callback.

That said I also find the design of GPIO callback API inadequate. We tend to enforce usage of CONTAINER_OF macro in places where its usefulness is questionable.

tbursztyka, 2018-12-10:
> We tend to enforce usage of CONTAINER_OF macro in places where its usefulness is questionable.

That's a false assumption. At the time it was designed it was fitting just fine with the known use-cases/requirements. If that does not fit anymore today, we will be all free to change this.

As described above, having to store a dev pointer in struct sht3xd_data is not going to bloat a struct for all gpio callbacks. I don't exactly see a problem with this, it seems to be a corner case.

If it was:
`typedef void (*gpio_callback_handler_t)(struct device *port,
					void *user_data,
					u32_t pins);`

You surely could pass the callback owner's dev pointer (and from there, getting back the context data, etc...) which would perfectly fit with the described use-case.

But the user_data pointer would still need to be somewhere, and per-callback, thus stored in struct gpio_callback. Growing it by 4 bytes. Which is not nice.

Or did I miss something ?

mnkp, 2018-12-10:
> But the user_data pointer would still need to be somewhere, and per-callback, thus stored in struct gpio_callback. Growing it by 4 bytes. Which is not nice.

Yes, that's true. I don't think there is a way around it. Current design of GPIO callback API which relies on CONTAINER_OF macro does save 4 bytes of RAM per callback if user does not need to access `dev->config->config_info` or similar data. That's unquestionably a very good argument to use it. However, in case of other APIs where we pass user_data to the callback we also need to grow the data structure in RAM by 4 bytes even if it's not always used. The resulting API is more intuitive and user friendly.

In any case my statement "I find the design of GPIO callback API inadequate" was premature. My apologies.

pabigot, 2018-12-11:
As a minor addition if we're going for consistency the order of parameters should be considered as well.

systemd uses the following standard for callbacks ([example](https://www.freedesktop.org/software/systemd/man/sd_event_add_time.html)):

* first parameter is a pointer to the object associated with the
  callback (here, the device pointer)
* next are any callback-specific parameters (times, channel ids)
* last is the `user_data` pointer

pabigot, 2018-12-13:
For watchdog I would want the user data to be a pointer to a noinit RAM object where I'd record information such as a stack trace showing what was executing when the watchdog occurred (I actually do this in my C++ Nordic framework).  Then that handler can be put in an application-independent library.

For GPIO, as I noted originally I can't get the information I need using `CONTAINER_OF` the GPIO callback and the GPIO device without "plumbing" that costs the same RAM as capturing it in the callback  object where anybody can take advantage of it.

I'm skeptical of claims that adding `sizeof(void*)` to the GPIO callback is a burden: how many GPIO callback structures exist in an application?  Probably only a couple in something that has sensors and a network connection (there are 33 declarations of a gpio_callback in the entire drivers directory covering all sensors, mostly one per device).  But it's true that it depends on the cost in RAM and ROM of the workarounds required when a generic pointer isn't available.

At a higher level I'm skeptical of arguments along the line of "I can't see why somebody might need this, so we won't support it."  It's not a great criterion for deciding the features of a generic framework/OS intended to support a wide variety of applications.

pizi-nordic, 2018-12-13:
In my opinion we should follow the simple rules:
- Every time we registering callback, we should be able to attach a pointer to user data.
- If callback comes from an subsystem, we should pass a pointer to given instance of the subsystem (driver, BT connection and so on).

I think this matches with the @pabigot proposal.

tbursztyka, 2018-12-13:
So the idea here is to fix something which is not broken, not to say that it is less memory efficient.

I am not against changes: the proposal would not raise the memory usage and would be performing better, it would be fine by me.

There is no reason to push for any convention on how callback in every API should be like.
We should stick to the case by case.

mnkp, 2018-12-17:
> Consider a driver for monitoring LiPo charging state that uses two external signals: (1) availability of external power, and (2) active charging status (which is valid only when external power is present). These signals share a handler function that executes on change to either signal to determine the state of the charging system and notify the application of that state.  [...]
>
> Now consider the possibility that a third board with these same signals happens to connect the two signals to two different GPIO instances, but with the same pin ordinal on each device (e.g. `P0.1` and `P1.1`). [...]
> 
> In the described situation `pins` doesn't allow you to tell which GPIO device is associated with the callback invocation, hence which interrupt has fired, hence which of the `gpio_callback` instances that are available through `user_data` is the one to reconfigure and which gpio device pointer to use for the reconfiguration.

That's a good argument in favor of having a callback provide both the device pointer and a generic pointer. However, in a real life cases I've seen where this situation had to be dealt with having a generic pointer only was never an issue. Here we have a situation where a single driver instance is using services of several instances of another driver (this could also be extrapolated on the application level). In such cases it is natural that every such driver instance would serve a different purpose. E.g. in the example above one gpio instance provides information about availability of external power the second about active charging status. As a result if any of the driver instances triggers a callback we need to execute a different piece of code. A better and cleaner design is to use different callbacks. If you use one callback only you'll have to put an if statement inside. Or worse a case statement if there are more scenarios to deal with.

> > Could you point me to a place in the code where such plumbing was necessary?
> 
> [...] my failure to find such a case wouldn't prove anything since we'd also have to assume that Zephyr's existing codebase covers the entire space of callback usage that might be envisioned by a device or application developer.

Surely, we need to design API that's most flexible and doesn't hinder the user. Yes, there might be cases we haven't thought of. But there will always be compromises to be made. The problem with passing both the device pointer and a generic pointer is that it's not entirely for free.

Looking at another production grade, open source project Zephyr heavily borrows from, namely Linux, there are very few API callbacks which pass both the device pointer and a generic pointer.

pabigot, 2018-12-18:
> However, in a real life cases I've seen where this situation had to be dealt with having a generic pointer only was never an issue.

This is a "real-life case" that applies to two different LiPo charger ICs used on Nordic-based boards (Thingy52 and Particle).  The signals are coupled: you can't correctly interpret the charging status unless you know the external voltage status.  The application should not be responsible for knowing that and implementing the necessary logic (I've done that; it's ugly).  We'll undoubtedly argue that more once we start defining a power source management API.

I don't really see a lot of architectural or process similarity between Linux and Zephyr, and had already pointed out systemd does pass both consistently, so I don't find that a compelling argument.

nordic-krch, 2018-12-18:
@mnkp i got convinced to have device and user data. There is rather no additional cost of having both. It's only added cost of user data and we know that we want to have that. Interrupt handler will anyway need to have access to device struct so it can pass it to the handler without a cost.

mnkp, 2018-12-18:
> @mnkp i got convinced to have device and user data.

That's one more voice in support of the proposal described in this issue. Surely, we will stick to the majority decision or what maintainers deem a better approach.

> There is rather no additional cost of having both.

I'll copy the reply I gave elsewhere

> It does come cheap. But - at least on ARM architecture - only up to 4 parameters are passed to the function as register variables. If the parameter list is longer we need to put them on the stack. So if we decide to have a general convention to always pass pointer to dev and to user_data there is a price to pay.

Having long parameter list makes code slower, always, for all users. If there is a corner case the API doesn't support well there is some overhead for this specific case. In my opinion making a per case decision on what callback signature should look like is a better approach. So that's one voice against the proposal described in this issue.

mnkp, 2018-12-18:
> This is a "real-life case" that applies to two different LiPo charger ICs used on Nordic-based boards (Thingy52 and Particle). The signals are coupled: you can't correctly interpret the charging status unless you know the external voltage status.

@pabigot I don't find this answer relevant to my previous argument of using two different callbacks. Maybe I misunderstood your problem or you my argument. Do you maybe have some code laying around you could point me to so I could analyze and understand your situation better?

pabigot, 2018-12-18:
No code that's self-contained enough to post here.  Torex XC680x chargers have an active-low Nch open drain charging state output.  The system is charging only if CSOn reads low and EXT_POWER reads active. If EXT_POWER is inactive CSOn will always be low but in that case it's not charging.  So to assess a state of "charging" you need both signals.  The natural place to do that is a single callback that is informed about change-of-state for both signals.

carlescufi, 2020-07-28:
API meeting 28th July:

- Agree to standardize on `struct device *` + `void *user_data` for all driver callback signatures, except those that might accept multiple callback registrations. This is the case for GPIO, where `CONTAINER_OF` is used instead for memory efficiency.

PR #27624 (2020-08-25): doc: document various best practices and expectations
This records the consensus from some recent API telecon discussions related to devicetree hardware vs software configuration, callback design, and how to ensure documentation is generated for conditional capabilities.

Closes #11976.
closed 2020-08-25 by carlescufi