zephyrproject-rtos/zephyr #11979: help to use zephyr-ninja flash

fghjhuang, 2018-12-10:
board: nucleo f401re
i can cmake the project basic/blinkly,then when i flash to the board the error open:
`david@ubuntu:~/Desktop/zephyr-master/samples/basic/blinky/build$ ninja flash`
`[1/1] Flashing nucleo_f401re`
`Using runner: openocd`
`Open On-Chip Debugger 0.10.0+dev-gea2753a7-dirty (2018-11-02-23:37)`
`Licensed under GNU GPL v2`
`For bug reports, read`
`http://openocd.org/doc/doxygen/bugs.html`
`Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD`
`adapter speed: 2000 kHz`
`adapter_nsrst_delay: 100`
`none separate`
`srst_only separate srst_nogate srst_open_drain connect_deassert_srst`
`Info : Unable to match requested speed 2000 kHz, using 1800 kHz`
`Info : Unable to match requested speed 2000 kHz, using 1800 kHz`
`Info : clock speed 1800 kHz`
`Info : STLINK v2 JTAG v23 API v2 SWIM v9 VID 0x0483 PID 0x374B`
`Info : using stlink api v2`
`Info : Target voltage: 3.252071`
`Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints`
`Info : Listening on port 3333 for gdb connections`
`    TargetName         Type       Endian TapName            State       `
`--  ------------------ ---------- ------ ------------------ ------------`
` 0* stm32f4x.cpu       hla_target little stm32f4x.cpu       halted`
`Info : Unable to match requested speed 2000 kHz, using 1800 kHz`
`Info : Unable to match requested speed 2000 kHz, using 1800 kHz`
`adapter speed: 1800 kHz`
`target halted due to debug-request, current mode: Thread `
`xPSR: 00000000 pc: 0x821f108c msp: 0x0000000c`
`auto erase enabled`
`Info : device id = 0x10006433`
`Info : flash size = 512kbytes`
`Error: timeout waiting for algorithm, a target reset is recommended`
`Error: error executing stm32x flash write algorithm`
`Error: flash write failed = 00000040`
`Error: error writing to flash at address 0x08000000 at offset 0x00000000`

`ERROR: command exited with status 1: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/david/Desktop/zephyr-master/boards/arm/nucleo_f401re/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /home/david/Desktop/zephyr-master/samples/basic/blinky/build/zephyr/zephyr.bin 0x8000000' -c 'reset halt' -c 'verify_image /home/david/Desktop/zephyr-master/samples/basic/blinky/build/zephyr/zephyr.bin 0x8000000' -c 'reset run' -c shutdown`
`run as "west -v ... flash ..." for a stack trace`
`

Any ideas?

jeevaknd, 2019-01-28:
how to install stlink drivers in ubuntu

fghjhuang, 2019-01-28:
you can see my blog->http://blog.smartlitehome.top/index.php/archives/59/
`sudo apt-get install libusb-1.0 //安装依赖库
git clone https://github.com/texane/stlink 下载
make release
make debug
cd build
cmake -DCMAKE_BUILD_TYPE=Debug ..
make
cd Release
sudo make install
最后一定要记得，把release目录下的st-flash文件拷贝到/usr/bin目录下才可以进行下载`

jeevaknd, 2019-01-28:
thank you 
最后一定要记得，把release目录下的st-flash文件拷贝到/usr/bin目录下才可以进行下载
cant understand please tell this command in english

fghjhuang, 2019-01-28:
ok, it means->remember to move the "st-flash" file from the dir->"release" to dir->"usr/bin/", then you can download to the board

jeevaknd, 2019-01-28:
we followed all the steps. Again we are getting the error

[0/1] Flashing stm32f429i_disc1
Using runner: openocd
Open On-Chip Debugger 0.10.0+dev-gea2753a7-dirty (2018-11-02-23:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v25 API v2 SWIM v14 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 2.893034
Error: init mode failed (unable to connect to the target)
in procedure 'init' 
in procedure 'ocd_bouncer'

ERROR: command exited with status 1: /home/hp/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/hp/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/hp/zephyr/boards/arm/stm32f429i_disc1/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /home/hp/zephyr/samples/hello_world/build/zephyr/zephyr.elf' -c 'reset halt' -c 'verify_image /home/hp/zephyr/samples/hello_world/build/zephyr/zephyr.elf' -c 'reset run' -c shutdown
run as "west -v ... flash ..." for a stack trace