zephyrproject-rtos/zephyr #11980: cmake: Build fails in an environment without `python` binary

yashi, 2018-12-10:
**Describe the bug**
The commit 9de5bc91093184bb1a57d031e76c089c29806d23 kills our build with `FATAL_ERROR` because the function `determine_python_version` can't find a binary called `python`.  Python 3 in some distro is named `python3`.  With some configuration, there is no alias to `python` or no Python 2 as `python`.  Thus it `FATAL_ERROR`.

**To Reproduce**
1. remove `python` binary.
2. build.

**Expected behavior**
Successfully build with and only with `python3`

**Impact**
Our CI failed.

**Screenshots or console output**
```
CMake Error at zephyr/cmake/backports/FindPythonInterp.cmake:46 (message):
  Failed to test python version
Call Stack (most recent call first):
  zephyr/cmake/backports/FindPythonInterp.cmake:58 (determine_python_version)
  zephyr/cmake/app/boilerplate.cmake:103 (include)
  CMakeLists.txt:3 (include)
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Native
 - Commit SHA or Version used: 9de5bc91093184bb1a57d031e76c089c29806d23

ulfalizer, 2018-12-10:
As a side note, CMake still chooses Python 3.4 for me, even though I have 3.6 and 3.7 installed as well.

ulfalizer, 2018-12-10:
Think I know what's going on:

The `python` executable is almost always Python 2 (but is Python 3 on Arch Linux I think).

Should probably test `python3` instead. Maybe that'll work on Arch as well. Otherwise, both might need to be tested.

SebastianBoe, 2018-12-10:
I have a fix, will push it soon.