zephyrproject-rtos/zephyr #11998: intermittent failures in tests/kernel/common: test_timeout_order: (poll_events[ii].state not equal to K_POLL_STATE_SEM_AVAILABLE)

andrewboie, 2018-12-10:
**Describe the bug**
```
starting test - test_timeout_order
thread 0 synced on timer 0

    Assertion failed at /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/tests/kernel/common/src/timeout_order.c:95: test_timeout_order: (poll_events[ii].state not equal to K_POLL_STATE_SEM_AVAILABLE)
```

**To Reproduce**
Seen only intermittently. Keeps happening in CI runs. Not sure how this can be induced repeatedly.

**Expected behavior**
Test case passes without incident.

**Impact**
Unclear. Might be a test case implementation issue. Might be a QEMU problem. Might be a subtle bug in the kernel that needs fixing.

pizi-nordic, 2018-12-11:
Looks like another issue in timer. Let me check...

pizi-nordic, 2018-12-11:
Unfortunately I cannot reproduce this...

andyross, 2018-12-11:
Oooh, good timing (because I'd already almost forgotten).  I stumbled on this one too in x86_64, then it went away when I fixed other bugs.  But this one I believe to be a bug in poll.

The relevant state being checked is that the poll event (the struct owned by the caller of k_poll() that returns information about what has happend) is only set in is_condition_met() in poll.c.  This gets called out of the loop at the top of k_poll() on entry to see what has already expired.

**BUT**, if k_poll() then goes to sleep and is woken up by the event being signaled, we do **not** travel through is_condition_met so the result state never gets set and k_poll() returns anyway.

That sounds fatal, but it's almost undetectable: poll is traditionally done in a loop: call it, see what changed and take appropriate action, and call it again.  So typical user code will just wake up, see that nothing has changed, do nothing, go back into poll and see the state variable the second time through.

The test, though, seems not to be calling poll in a loop, so it will be sensitive to timing issues depending on whether k_poll() gets through its initial check before or after the event is signaled.

I'm a little crushed for the next day or three, but if someone else has time to dig before I get to it, I'll put $20 down that this is the root cause.  Otherwise just assign it to me :)

andrewboie, 2018-12-12:
@andyross I have some time today, I'll take a look

andrewboie, 2019-02-15:
> **BUT**, if k_poll() then goes to sleep and is woken up by the event being signaled, we do **not** travel through is_condition_met so the result state never gets set and k_poll() returns anyway.

The object-specific handle_poll_events seems to do this for us. Otherwise this test would never work, the k_poll() call in test_timeout_order() always blocks.

The problem AFAICT is priority setting; the test main thread wants to lower priority to let the child threads all run, but instead of setting to a priority lower than the child threads, it gets set to the SAME priority, which means the k_poll() call may complete when some of the child threads have done their thing, but not all of them.

andrewboie, 2019-02-15:
Notice output:

```
===================================================================
starting test - test_timeout_order
thread 0 synced on timer 0

    Assertion failed at /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/tests/kernel/common/src/timeout_order.c:95: test_timeout_order: (poll_events[ii].state not equal to K_POLL_STATE_SEM_AVAILABLE)

FAIL - test_timeout_order
===================================================================
```

Only thread 0 sychronized on the timer. We expect to see:

```
===================================================================
starting test - test_timeout_order
thread 0 synced on timer 0
thread 1 synced on timer 1
thread 2 synced on timer 2
PASS - test_timeout_order
===================================================================
```

This looks like another annoying QEMU icount thing; threads 1 and 2 just never got a chance to release their semaphores. Trying to see if there is some way I can make this more robust.

andrewboie, 2019-03-07:
@gnuless I would sync with @andyross once you start looking at this.

andyross, 2019-03-08:
We can close this one.  This test is very reliable on all platforms under the various workarounds detailed in #12553.  There's nothing to do here in particular.

closed 2019-03-08 by andyross