zephyrproject-rtos/zephyr #12003: Device Idle Power Management

ramakrishnapallala, 2018-12-11:
In Zephyr, we support system power management and as part of the system-wide PM or CPU idle PM we
have mechanisms to suspend devices while entering deep sleep states. But to save power at the more granular level we can turn off or suspend the devices which are not active irrespective of the CPU state and bring them back when the need comes. The concept is similar to the runtime PM implementation on Linux. 

We can introduce a simple framework to track the device usage count and based on the usage count we can put the devices into suspend state. The runtime PM core tracks the usage count of all the devices and puts the devices into either suspend/idle/active states depending on the device current state.

ramakrishnapallala, 2018-12-11:
FYI @nashif @carlescufi

carlescufi, 2018-12-11:
@ramakrishnapallala thanks for creating the issue. Copying @pizi-nordic since he's the power management expert on our side.

pizi-nordic, 2018-12-11:
I fully support this idea, however I have few concerns how to handle peripheral power state.

Some peripherals could be easily managed: For example SPI master, I2C master and so on. When peripheral is not used we  can simply turn it off (the latency of turning on might be the only concern here). For most of the drivers this is already implemented at driver level.

However peripherals like UART or SPI slave cannot be turned off in runtime unless we know that they will be not used by external world. And this knowledge is not always available on system level, as behaviour depends on the application. So for such cases we will need special API.

Last but not least: The latency is IMHO a bigger issue in Zephyr-class devices than in big OSes. We will need infrastructure for defining interrupt/device response latency in order to choose best power mode satisfying real-time requirements of application.

ramakrishnapallala, 2018-12-12:
Thank you @pizi-nordic for your inputs. I will keep these things in mind while defining the API's.

mnkp, 2019-03-16:
Since #13382 was merged should we close this issue?

pizi-nordic, 2019-03-18:
I think so.

closed 2019-03-18 by pizi-nordic