zephyrproject-rtos/zephyr #12025: OS Pwr Manager doesn't put nrf52 into LPS_1

bearsh, 2018-12-11:
**Describe the bug**
Using a very simple app (`while (1) {k_sleep(MSEC_PER_SEC);}` in main thread only) with os pwr manager enabled, the lowest current consumption I get is around 700uA, which looks more like LPS_0 than LPS_1

**Expected behavior**
A sleeping SoC for almost 1s with an expected current consumption of around ~5uA rather than 700uA

**Additional context**
Running the pwr manager demo, the systems *sometimes* (not reliable) enters LPS_1.

tautologyclub, 2019-01-25:
If you read the code for the power management example it's pretty clear this isn't a bug. You have to implement the suspend hook yourself. Suggest closing.

bearsh, 2019-01-25:
@tautologyclub do you mean [pwer_mgr](https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/subsys/power/power_mgr)? I can't find any reference to a hook I have to implement. Just to make it clear, this issue is about 'OS Power Management' which should enter the appropriate sleep state 'by itself', see also [power.c](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/power/power.c)

pizi-nordic, 2019-03-26:
There are no low power states in nRF5x SoCs. The https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/subsys/power/power_mgr example introduces them artificially in order to show the OS-based policies.

There is no magic switch which reduces power consumption to XX uA in your design.
And there never will be one.

bearsh, 2019-03-26:
@pizi-nordic ok, let's not name them LPS_x (as I just notived, commit f04a4c9deb55400571fd567c8ed1378015e18a24 removed that naming). but without any magic switch, running the simple application mentioned in the description and have the pm handled by the kernel (logging etc. disabled) shouldn't end up in a current consumption of 700uA.

carlescufi, 2019-08-26:
@pizi-nordic @pabigot is this still a valid issue?

pabigot, 2020-08-24:
A year later I don't think this is worth keeping open.  Currently controlling power is mostly manual, and the referenced sample has been replaced by `samples/boards/system_off` which shows (crudely) how to do it for Nordic.  At some point this will be done automatically, but it's an ongoing process being tracked elsewhere (e.g. entry point at #24228).

closed 2020-08-24 by pabigot