zephyrproject-rtos/zephyr #12029: CONFIG_STACK_CANARIES doesn't work on x86_64

andyross, 2018-12-11:
This feature isn't working right now on x86_64, and the problem seems deep.  Generated code via -fstack-protect arguments panics very soon after boot.  I worry this may be an ABI issue with support for x32, but it needs investigation.  The tests are whitelisted for now.

nashif, 2018-12-12:
you are not hitting this 95936e2b1433b8aaac798861050a0353149fc0a4, are you?

andyross, 2018-12-12:
Ooh, maybe.  That could totally produce the symptom's I saw, though I'm pretty sure it persisted as I rebased.  Honestly I don't know anything about this gcc feature and just punted once I realized the failure was just that canaries don't work even on hello_world.  Will check.

andrewboie, 2018-12-12:
if you hit that problem you'll see references to memory in %fs or %gs segment in the beginning of functions.

andyross, 2018-12-12:
Hah, yeah, that was it.  I thought I'd added that whitelist patch after my final rebase, but I guess not.  It works on top of current HEAD.  Just pushed a rebased PR.  That was easy, thanks.

closed 2018-12-12 by andyross