zephyrproject-rtos/zephyr #12037: nrf52840 ram retention example failed

Jiurifei, 2018-12-12:
Greetings,
I want to  use a global variable to store a status value .  It can be reserved even if I execute reset operation. I referred to  nRF52840 documentary, and I got following information:
The RAM power control registers are used for configuring the following:

SebastianBoe, 2018-12-12:
You might want to use a NRF_POWER-GPREGRET[] variable or a flash location instead.

Anyway, probably, something is clearing the variable during the boot procedure. You need to make sure it is not placed in a section that is modified during the boot procedure, e.g. it should not be in the bss or data section.

Jiurifei, 2018-12-13:
@SebastianBoe Thank you for your suggestion. In fact we don't gvie top priority to flash storage. From your advice, I have thought that ram control strategy is divided into two aspects: hardware and software. operating register is the operation at hardware layer. And if  I do nothing at software layer, the ram can also change, e.g, reset. Am  I right？ please let me know.

SebastianBoe, 2018-12-13:
It's difficult to understand what you are saying, but it sounds about right.

SebastianBoe, 2018-12-28:
Closing as resolved.

closed 2018-12-28 by SebastianBoe