zephyrproject-rtos/zephyr #12039: Clock control API should allow callback to be specified

masz-nordic, 2018-12-12:
When starting the clock asynchronously, there is no way to specify an additional callback function to propagate clock events to application.

API should include an optional registration of a handler that would be called when clock event (for example `HF_CLOCK_STARTED`) is triggered.

cvinayak, 2018-12-14:
@carlescufi @pizi-nordic I agree on the requirement, but this needs API discussion and porting across various architectures and SoC from multiple vendors.

pizi-nordic, 2018-12-14:
@cvinayak: That is true. The implementation must be generic. I will try to raise this topic on API meeting.

carlescufi, 2019-01-07:
@pizi-nordic maybe you could propose an (optional) API that we could use, implementing it initially only on Nordic devices?

pizi-nordic, 2019-01-08:
@carlescufi: OK.

PR #16675 (2019-06-25): drivers: clock_control: API extension proposal
Proposal which extends API to allow asynchronous clock enabling.

Fixes #11107.
Fixes #12039.
closed 2019-06-25 by carlescufi