zephyrproject-rtos/zephyr #12040: Log messages are dropped even with huge log buffer

jhedberg, 2018-12-12:
**Describe the bug**

There's something seriously wrong with dropped log messages, at least when using the shell. I've configured my app (tests/bluetooth/mesh_shell) to have a 32kB LOG_BUFFER_SIZE, 64 log_strdup buffers, and increased the FIFO size in subsys/shell/shell_uart.c from 10 to 100, yet I get the following:
```
***** Booting Zephyr OS zephyr-v1.13.0-2680-g523acef71c *****
Press the <Tab> button for supported commands.
Before any Mesh commands you must run "mesh init"

uart:~$ mesh init
Bluetooth initialized
Mesh initialized
Use "pb-adv on" or "pb-gatt on" to enable advertising
[00:00:02.042,114] <dbg> bt_mesh_settings.mesh_set: s
[00:00:02.042,175] <dbg> bt_mesh_settings.mod_set: Decoded mod_key 0x0003 as elem_idx 0 mod_idx 3
[00:00:02.042,175] <dbg> bt_mesh_settings.mod_set_sub: val (null)
[00:00:02.042,175] <dbg> bt_mesh_settings.mod_set_sub: Cleared subscriptions for model
[00:00:02.042,297] <dbg> bt_mesh_settings.mesh_commit: sub[0].net_idx 0xffff
[00:00:02.042,480] <inf> bt_hci_core.bt_dev_show_info: Identity: eb:8f:fb:67:e8:ee (random)
[00:00:02.042,480] <inf> bt_hci_core.bt_dev_show_info: HCI: version 5.0 (0x09) revision 0x0000, manufacturer 0x05f1
[00:00:02.042,480] <inf> bt_hci_core.bt_dev_show_info: LMP: version 5.0 (0x09) subver 0xffff
uart:~$ shell stats show
Lost logs: 39
uart:~$ 
```

So 8 messages make it through to the console, while 39 were dropeed. How is that possible with my configuration? Here's the full diff of changes I've done:
```
diff --git a/subsys/bluetooth/host/mesh/settings.c b/subsys/bluetooth/host/mesh/settings.c
index 60630e2600..973bba5e9f 100644
--- a/subsys/bluetooth/host/mesh/settings.c
+++ b/subsys/bluetooth/host/mesh/settings.c
@@ -672,6 +672,8 @@ static int mesh_set(int argc, char **argv, void *val_ctx)
                return -EINVAL;
        }
 
+       BT_DBG("%s", log_strdup(argv[0]));
+
        for (i = 0; i < ARRAY_SIZE(settings); i++) {
                if (!strcmp(settings[i].name, argv[0])) {
                        argc--;
diff --git a/subsys/shell/shell_uart.c b/subsys/shell/shell_uart.c
index 4e8840267e..6aa464f6ad 100644
--- a/subsys/shell/shell_uart.c
+++ b/subsys/shell/shell_uart.c
@@ -21,7 +21,7 @@ LOG_MODULE_REGISTER(shell_uart);
 SHELL_UART_DEFINE(shell_transport_uart,
                  CONFIG_SHELL_BACKEND_SERIAL_TX_RING_BUFFER_SIZE,
                  CONFIG_SHELL_BACKEND_SERIAL_RX_RING_BUFFER_SIZE);
-SHELL_DEFINE(shell_uart, "uart:~$ ", &shell_transport_uart, 10,
+SHELL_DEFINE(shell_uart, "uart:~$ ", &shell_transport_uart, 100,
             SHELL_FLAG_OLF_CRLF);
 
 #ifdef CONFIG_SHELL_BACKEND_SERIAL_INTERRUPT_DRIVEN
diff --git a/tests/bluetooth/mesh_shell/prj.conf b/tests/bluetooth/mesh_shell/prj.conf
index 9a63489414..4f5c93c924 100644
--- a/tests/bluetooth/mesh_shell/prj.conf
+++ b/tests/bluetooth/mesh_shell/prj.conf
@@ -66,3 +66,9 @@ CONFIG_BT_DEBUG_LOG=y
 
 CONFIG_BT_MESH_IV_UPDATE_TEST=y
 CONFIG_BT_MESH_DEBUG=y
+
+CONFIG_BT_MESH_DEBUG_SETTINGS=y
+
+CONFIG_LOG_PROCESS_THREAD_STACK_SIZE=2048
+CONFIG_LOG_BUFFER_SIZE=32768
+CONFIG_LOG_STRDUP_BUF_COUNT=64
```

**To Reproduce**
Steps to reproduce the behavior:
1. apply the above diff
2. build tests/bluetooth/mesh_shell for an nRF52840 board
3. run "mesh init" and "mesh provision" in the shell
4. power cycle the board (wait a second or two after "mesh provision")
5. run "mesh init"
6. run "shell show stats" - observe that already some messages have been dropped
7. run "mesh reset"
8. power cycle the board
9. run "mesh init" and "shell show stats" - you should get something similar to the console output that was pasted in the beginning of this description.

jhedberg, 2018-12-12:
@jukkar @tbursztyka FYI - I heard you had also had some issues with logging reliability

nordic-krch, 2018-12-14:
The reason for logs being dropped is that logger thread gets burst of messages (>300 in that case). Sends them to all active backends (shell). Shell thread is waken up but since it's on the same priority it pends. As shell queue gets full, it starts dropping messages.

There is also a bug in shell that size of the fifo does not come from the macro parameter but from global config ( `SHELL_MAX_LOG_MSG_BUFFERED`).

I don't have yet good solution for that. One option that can be added is to call k_yield() in logger thread after message is sent to backends. That will trigger log processing in threads with the same priority like shell but does not solve that case. That is because shell thread will pend on event from uart interrupt so logger thread will continue pushing messages. Another solution would be to change shell reaction to fifo full. It could block logging thread until fifo has room but need to think if that can be dangerous.

nordic-krch, 2018-12-14:
My idea is to add option to shell if we want to have reliable logs or not in that shell. In future, we might have shell over other transports (bluetooth, thread, network, etc.) and in that case we might want to drop logs if transport is too slow for given amount. On the other hand fast, debugging transports (uart, rtt) are good candidates where we might want to force reliable logging. 'Reliable shell' would block logger thread until there is a room in the queue. As logger thread is blocked by one backend, irresponsible use might block logs in the system - that's the risk.

@jhedberg asked if thread priority may be increased. Initially, i thought that it's also applicable solution but actually it is not. Shell thread will still be limited by the transport so when it pends waiting for event from transport, logger thread will pass new messages.

jhedberg, 2018-12-14:
@nordic-krch that sounds like a good option. Let me know when you have something testable for it.

PR #12125 (2018-12-19): shell: Improve handling of log messages
If burst of log messages was passed to the shell log
backend, it was likely that messages were lost because
shell had no means to control arrivals of log messages.

Added log message enqueueing timeout to the shell instance
to allow blocking logger thread if short-term arrival rate
exceeded shell capabilities.

Added kconfig option for setting log message queue size
and timeout in RTT and UART instances. Added section in
shell documentation which explains interaction between
the logger and shell instance acting as a logger backend.

Fixes #12040.
closed 2018-12-19 by carlescufi