zephyrproject-rtos/zephyr #12044: [Question] [Blutooth Mesh] How proxy client create the proxy PDU

gofortime, 2018-12-12:
Hi Johan,

Here is a question about the Bluetooth Mesh proxy function. I am trying to do some test on the proxy function of the Mesh stack. First, I have a device setup as proxy server (acts as provisionee, and has been provisioned in a mesh network). Then I tried to setup a proxy client, which could connect to proxy server device by discovering proxy GATT services/characteristics. And then I tried to write a proxy PDU to the proxy server via the "Mesh Proxy Data In" characteristic. But what I am struggling is the content of the PDU. According the Mesh spec, when PDU arrives on proxy side, its content is encrypted and obfuscated (spec 3.8.7 section). So, on the proxy client side, I have to do the encryption and obfuscation. I am just wondering how the proxy side know the information, including IV index, privacy key, nonce, which is required to encrypt and obfuscate the PDU? Is there any sample code that I can refer to to setup a proxy client?

Here is a summery of my question:
1. How does proxy client know the IV index, nonce information which is required to encrypt and obfuscate the proxy PDU?
2. Is there any sample code to setup a proxy client node?

Thanks for your reading, and look forward to your response.

Thanks,
Vincent

jhedberg, 2018-12-12:
1. The proxy client is just another Mesh network node, and knows these values the same way as any other node knows them. I.e. it gets the initial values from a provisioner and then updates the IV Index whenever the proxy server indicates that there's been an update to it.

2. I wonder if Nordic's nRF Mesh app is Open Source since it's a proxy client. @carlescufi do you know? Another possibility is to look at meshctl from BlueZ, since that's a proxy client as well.

carlescufi, 2018-12-12:
Regarding nRF Mesh and open source, I think @philips77 and @trond-snekvik should be able to answer that.

Jiurifei, 2018-12-13:
I think you may hard code the input data if necessary. Anyaway, mesh proxy function is just like a pipe. So, only thing you need to do  is to get network encryption data produced by bt_mesh_send_net and use write without response function to send it. Also, you may have to change some judging conditions in proxy server project. That's just my sudden idea, it may not work.

roshanrajaratnam, 2018-12-13:
@gofortime 
1. IV Index is a property of the network. So the proxy client that does the provisioning needs to know and store the properties of the network. Nonce is something that is generated on the fly depending on the sequence number of a particular pdu, iv index  and other parameters. 

2. Below is the links to the libraries on both Android and iOS

nRF Mesh for Android and iOS are on GitHub:
https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/
Both libraries have sample apps that are published on stores.
CC @mostafaberg

If you have more questions get in touch :)

gofortime, 2018-12-14:
@jhedberg @carlescufi @Jiurifei @roshanrajaratnam 

Hi guys, thanks for your kindly replies. Let me examine some details first according to your suggestions. Thank you again.

Thanks,
Vincent

carlescufi, 2019-08-15:
Question answered

closed 2019-08-15 by carlescufi