zephyrproject-rtos/zephyr #12045: frdm_k64f + shell breaks Ethernet driver

tbursztyka, 2018-12-12:
**Describe the bug**
As soon as the shell is being used, for some reason it does break the ethernet driver on frdm_k64f.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/net/dhcp4_client
2. add CONFIG_NET_SHELL=y in prj.conf
3. mkdir build; cd build
4. cmake -DBOARD=frdm_k64f
5. make
6. Put the result zephyr/zephyr.bin into the board (MBED drive)
7. via the console, once dhcpv4 has successfully finished, ping the gateway: it will fail. Not also this:

net ping 192.168.1.1
Sent a ping tSento 192.168.1.1

See the "tSento" ? 

The ping will timeout.

a net arp command will show a proper entry for the gateway.
do a : net arp flush
re-ping the gateway
another net arp command will show empty table.
That's because it could not receive the ARP reply for address lookup. The ethernet device seems to receive nothing

Note that wireshark shows that packets are still going out of the board.

**Expected behavior**
ping should work
ethernet device should not break on reception

**Additional context**
I did a git bisect, and the culprit is: commit 10aeb0e8ab1849ddc51c663ad591eb937ff20f87
If you unset  CONFIG_SHELL_BACKEND_SERIAL_INTERRUPT_DRIVEN, all works again.

So something is broken in either the shell or the uart driver of the board.

qianfan-Zhao, 2018-12-12:
A same question:  #11281

tbursztyka, 2018-12-12:
Indeed, I remember now that issue. It's not solved yet? :/ 

Anyway, closing that one as duplicate.

closed 2018-12-12 by tbursztyka